martin-ger/esp_wifi_repeater #76: Very low speed on old esp batch

imayoda, 2017-09-11:
Hello there again, 
I've tried to use the stock (old sdk and new) on an old (2015) esp-03 with 512K flash... it somewhat works but when it comes to bandwidth it tops at 0.3MB/s ... tried everything, and circuit is in good order.
On the nodemcu 0.9 it's perfect and gives, on single client connection, a whooping 8-9Mbit/sec .. Amazing !
The slow esp03 gives 1/1 bandwidth in fritz!box wireless stations screen, while the node tops at 54/32 sometimes, but never 1/1 constantly like the old esp... (tried in every signal strength, low and high, the same)
Any suggestions?

Another thing that happens on the old esp is that the startup open AP comes in sight but very hardly gives an IP and sometimes is very hard to get to the config page.. Could it be that Serial writing is somewhat interfering with wifi functions? I noticed the fast reconnection message happening in serial console.. once it connects to the SSID, the AP will give connection to clients very easily.
Mistery is getting thick.
Kudos

martin-ger, 2017-09-12:
Don't think, this is a software problem. While I never tested an ESP-03 modul, I have experience with a Sonoff switch, that also has 512KB mem only. This is not a problem.

You might try to "set speed 160" to increase the CPU speed, but I assume this will have no impact. My guess would be, some problems with the antenna: the ESP-03 has the ceramic antenna, right? It might loose packets or drop down to low transmission speeds. You might try to sniff with wireshark on a linux box the connection to the ESP to verify this.

imayoda, 2017-09-12:
Roger that, never thought ceramic antennas could be so bad, in fact it drops a lot of packets, sometimes resulting in incomplete html pages etc.
Could be that esp03 modules have a double antenna possibility, and the ANT pin is wired somewhat long on the breakout board, shifting the main ceramic antenna freq. reso.
I'll return on this as I have a esp07 module, another one with ceramic ant. and a connector wired in parallel. It will throw some light on this mistery for sure.
Thanks a lot in the meanwhile :)

imayoda, 2017-09-14:
I've made a repeater with esp07 8Mbit flash , and both with ceramic or external antenna (cutting out the ceramic) it performs like the slow 03 module. Same identical results
In serial console I often get LmacRxBlk:1, what is it?
I'm out of clues

imayoda, 2017-09-15:
Got it ... after some scope investigation I've found out that there were some voltage transients in the power supply (after the regulator) so I added a 1000uF 10V low esr cap to smooth it out and now IT WORKS as it is meant to.
Have to say that I have developed many esp protos during last two years, and my standard approach never betrayed me.. must be that heavy traffic requires heavy capacitor (before I was installing 47-100uF smd caps).
The most incredible thing is that actually ESP manages somehow to keep up working in reduced bandwidth (1/1Mbit) mode even with insufficient power supply.. magic :)
Kudos

closed 2017-09-15 by imayoda