martin-ger/esp_wifi_repeater #77: Feature required.....

chathurangawijetunge, 2017-09-12:
Is DDNS update possible.
If not it would be very nice to have it

martin-ger, 2017-09-12:
What do you think of? Registering the external IP address at some internet DDNS server? Does it make sense? The ESP has no PPP interface. The external IP address is assigned via an DHCP server which is typically itself a WiFi NAT-Router. Thus it is usually not the public IP address from the provider, but in itself a private IP. The router that has the external IP should register that at an DDNS service and most WiFi home routers have that feature.

If the ESP does the job, what protocol do you think of? DDNS with RFC 2136 is somewhat more work. DDNS via an HTTP request is fairly simple, but the URL format depends on the service provider: see e.g. https://github.com/ayushsharma82/EasyDDNS

chathurangawijetunge, 2017-09-13:
You are write..... but most branded roughers don't offer many free DDNS updates.
I use simple get requests.
I thought it would be nice to have it with ur nat rougher.