martin-ger/esp_wifi_repeater #79: Problem loading google on phone

gthb96, 2017-09-19:
Hello,
When I try to make a google search sometimes the loading just gets stuck at the beginning.
Sometimes I also get an error from Chrome "err_quic_protocol_error".

Has anyone experienced something similiar?
[image]

martin-ger, 2017-09-20:
Interesting - have no experience with QUIC. Andriod Firefox thinks, it is a good idea to flood the AP with hundreds of small empty packets before starting TCP, this sometimes leads to an overflow, but TCP/HTTP recovers from that. Perhaps we have a similar problem here?

Maybe you could try to enable monitoring and get a trace of a successful and a failed conversation. Later this week I might find some time to look into that.

gthb96, 2017-09-20:
What do you mean by enable monitoring?
I would like to make a trace for you from successful and failed.
I already tried to wireshark this but couldn't find anything useful there.

gthb96, 2017-09-20:
I have some dumps for you.
There are two were it hangs forever and never loads.
There are two successful ones where it loads normally.
There is one were it hangs for 15 seconds and finishes loading afterwards.

File link: http://rgho.st/8SJHfzWzL

martin-ger, 2017-09-22:
Thank you for sending me the traces. After looking into them and reading a little bit about QUIC I do understand that some QUIC (UDP) connections work even in the unsuccessful traces and others seem to fail. I don't see an obvious reason, but this is hard anyway as nearly every other info than CID and Seq-No is encrypted.
So my guess is, it has something to do with UDP NAPTing. UDP NAPTing is tricky anyway as it has no defined end of the protocol interaction. This means: when to delete a NAPT-entry for UDP? The only viable option is a timeout. The current value for this is 2 seconds - after 2 seconds of inactivity the NAPT-mapping is closed, i.e. the "connection" interrupted. This is fine for a quick DNS lookup, but quite short if you build TCP-style connections on top of it. In comparison: open TCP connections have a timeout of 30 min, closed TCP connections still 20 secs.

Of course this is a trade-off: the table of NAPT-entries is limited and long timeouts would occupy a lot of space.

A try to fix the problems would be to increase the 2 seconds to something higher (20 secs?) in
line 20 of " esp-open-lwip/include/lwip/lwip_napt.h":
#define IP_NAPT_TIMEOUT_MS_UDP (2*1000)

Do you have the development environment to test this or should I provide you with a patched version of the liblwip_napt.a?

gthb96, 2017-09-22:
If you could provide me the patched version it would be nicer than installing the environment.
Thanks.