martin-ger/esp_wifi_repeater #80: Compile with USER_GPIO_IN not working

harryd100, 2017-09-22:
First of all, thanks for this great piece of SW, really like it.

I got everything working except when I compile with USER_GPIO_IN enables (12), I get:

user/user_main.c: In function 'gpio_intr_handler':
user/user_main.c:2190:6: error: passing argument 3 of 'ets_timer_setfn' makes pointer from integer without a cast [-Werror]
      os_timer_setfn(&inttimer, int_timer_func, easygpio_inputGet(USER_GPIO_IN));
      ^
In file included from user/user_main.c:4:0:
/opt/esp-open-sdk/sdk/include/osapi.h:67:6: note: expected 'void *' but argument is of type 'uint8_t'
 void ets_timer_setfn(os_timer_t *ptimer, os_timer_func_t *pfunction, void *parg);
      ^
cc1: all warnings being treated as errors
Makefile:140: recipe for target 'build/user/user_main.o' failed
make: *** [build/user/user_main.o] Error 1

I'm not a real C guru, so maybe somebody can tell me why this is not working? It looks like ets_timer_setfn is expecting a void, but easygpio_inputGet returns an uint8_t, but I don't know how to correct this

regards, Harry

harryd100, 2017-09-22:
Did some more investigation. void *parg is not used in int_timer_func, so os_timer_setfn(&inttimer, int_timer_func,  easygpio_inputGet(USER_GPIO_IN)) could be replaced by os_timer_setfn(&inttimer, int_timer_func,  NULL), right?

martin-ger, 2017-09-22:
Strangely, for me it compiles fine even without modifications. But you are right, "os_timer_setfn(&inttimer, int_timer_func, NULL)" is fine as well and the type is correct. Will change that.

closed 2017-09-22 by martin-ger

harryd100, 2017-09-22:
Maybe this is because i use SDK 2.1?

martin-ger, 2017-09-22:
Probably - should install this soon....