martin-ger/esp_wifi_repeater #82: Secure MQTT (mqtts)

m-soori, 2017-09-25:
do you support mqtts too or only not secure mqtt on port 1883 ??

martin-ger, 2017-09-25:
Only unencrypted MQTT. The used lib has the feature, but certificate handling via the CLI this is cumbersome. If you want it, some changes in the connection setup in user_main() are needed.

MQTT over WiFi to a local broker is safe, as the links are individually encrypted. Only MQTT to a Broker in the Internet is critical.

m-soori, 2017-09-25:
thanks alot
can explain for me how you encrypt link on local wifi?

martin-ger, 2017-09-25:
WPA2 security encrpyts all WiFi-frames. It is "only" based on a shared password - thus only useful in a home environment where everything is under the control of one admin. But at least link encryption is provided and to my knowledge still considered as strong enough.

closed 2017-12-31 by martin-ger