martin-ger/esp_wifi_repeater #83: Figure request : Wish to know list of station's hostnames that connected to the repeater.

TridentTD, 2017-09-28:
Can the esp_wifi_repeater know the list of station's
hostnames, mac & ip & dhcp lease time ?

Thank you.

martin-ger, 2017-09-29:
"show stats" gives you the IPs and MACs of all stations that are connected (or have been asking the DHCP server). Names are difficult, as there is no DNS server and Station names in DHCP are not recorded (as far as I have seen up to now). Lease times can be extracted from DHCP, requires some work. Will look into that.

TridentTD, 2017-09-29:
Thank you for answer.

Is there any way to know station's hostname from esp_wifi_repeater?

harryd100, 2017-10-15:
A 'normal' DHCP client sends it's initial hostname in DHCP option 12 during the DISCOVER/REQUEST.

martin-ger, 2017-10-15:
I think the lwip DHCP client of the ESP tries to send the nif->hostname set by "wifi_station_set_config()" to the DHCP server - at least for me with my home router (an old FritzBox) this does not work (the FritzBox doesn't know the name while it does from Windows, Android. or IOS boxes).