martin-ger/esp_wifi_repeater #84: Wifi created by esp8266 doesn't show up

mycodedstuff, 2017-09-29:
After flashing **esp8266** and booting it up the **nodemcu** led starts flashing but the wifi which it has to create doesn't show up.
I have followed the exact same instructions which were provided in the markdown.

martin-ger, 2017-09-30:
What does the serial console show?

hemangjoshi37a, 2017-10-05:
same problem.
When I flashed it it showed no wifi or serial right after the flashing, but when I reset I got this error on serial 

```
epc1=0x40240a08, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
```
and still no AP or wifi showing up.

Please help.

martin-ger, 2017-10-05:
I am pretty sure, either your binary is corrupted or your power supply should bei stabilized.

hemangjoshi37a, 2017-10-05:
Okay then how to solve it?

hemangjoshi37a, 2017-10-05:
Actually  I read the whole error and it looks something like this. Please take a look at this.

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30884, room 16 
tail 4
chksum 0xc3
load 0x3ffe8000, len 2060, room 4 
tail 8
chksum 0x7e
load 0x3ffe8810, len 8308, room 0 
tail 4
chksum 0x3b
csum 0x3b
Fatal exception (0): 
epc1=0x40210cd0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40210cd0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40210cd0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40210cd0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40210cd0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40210cd0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40210cd0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40210cd0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40210cd0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40210cd0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 

```

hemangjoshi37a, 2017-10-05:
@martin-ger sir,
If the firmware would be corrupted then the programm would not run itself on the MCU because it is made of 1s and 0s. So Bin file I think is okay. And as far as power supply is concerned I tried to turn on the MCU using my 2 ampere power supply so I think power supply might not be a issue as well.

hemangjoshi37a, 2017-10-05:
Actually I have two different made NodeMCUs and I tried with both but the same result as mentioned above.

hemangjoshi37a, 2017-10-05:
Is it possible from your side to generate the bin files solving this bug?

martin-ger, 2017-10-05:
This firmware has been online for weeks and it works for many users. Don't think the repository here on github is corrupted. You might try the firmware_1.5.4 as alternative. Or older versions of this repository. All known to be good for many users. 

Try to check  the check sha1sums of your diwnloaded files. Guess the download failed. (ASCII Mode?) Or try to build it yourself with the esp_open_skd.

hemangjoshi37a, 2017-10-05:
Okay sir.  I will try that.

bonnefoi, 2017-10-09:
If you are unable to flash correctly your ESP8266, you can try to use the sub option --flash_mode dio of the option write_flash and use esptool v2.1. Works on my Wemos.
Hope this helps.

hemangjoshi37a, 2017-10-10:
Actually I solved this issue by using different flash memory size. Previusly I was using 4mb but then I realised that the picture shows to set 8mb so I tried it and it worked fine. I was setting 4mb flash size because my device has 4mb flash but it is weird and true that setting 8mb actually works fine. But Thax guys @martin-ger @bonnefoi for help.

closed 2017-10-10 by martin-ger