martin-ger/esp_wifi_repeater #85: It´s not possible to create or to connect to a ssid with space separated characters

brunooliva, 2017-10-03:
If you try to create a ssid like "A B C" or connect to a ssid like "A B C" by console it will save only the first letter. 
If you try to do the same described above via web interface the "space" is substituted by a plus sign "+".

guiyti, 2017-10-03:
This is a banal problem, because your B and C letter are interpreted as arguments, so if you want a space caracter on ssid you need to use ASC II codes. In your example you need to create a ssid named as "A%20B%20%C". Banal.

brunooliva, 2017-10-03:
Thank you very much guiyti, it worked!

closed 2017-10-03 by brunooliva

zawmintun1, 2018-01-02:
I tried on Telnet Android apk like Tech Zaw  as Tech%20Zaw but I cannot fix. Please help how to do on Telnet

zawmintun1, 2018-01-02:
Thanks all ,,,Now my job done by \space as(Tech\ Zaw).