martin-ger/esp_wifi_repeater #86: Not running on a NodeMCU V3

mailmartinviljoen, 2017-10-17:
Loaded 0x00000.bin and 0x00001.bin on a NodeMCU v3 with the ESP flashtool. Power cycled it by unplugging the USB cable. When monitoring the COM Port with putty I get a message.

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

ets_main.c
```

double tripple checked the flashtool settings are correct

martin-ger, 2017-10-18:
It is 0x10000.bin (not 0x00001.bin) and the respective address - guess you made this right? Did you try the 1.5.4 Firmware

mailmartinviljoen, 2017-10-18:
I'm sure its a typo in my original message. Will most certainly try again tonight :-) ,Have not tried the 1.5.4 firmware. If all fails i have a couple of ESP12s that I can  salvage from an old project .

mailmartinviljoen, 2017-10-18:
I can confirm my original message had a type i was supposed to write 0x10000 instead of 0x00001

I tried the 1.5.4 firmware aswell getting a different message using1.5.4

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 31148, room 16
tail 12
chksum 0xef
ho 0 tail 12 room 4
load 0x00000000, len 0, room 12
tail 0
chksum 0xef
load 0x00000000, len 0, room 4
tail 0
chksum 0xef
csum 0xef
csum err
ets_main.c

also tried 0x10000 ass opposed to  0x40000

mailmartinviljoen, 2017-10-18:
Can also confirm the NodeMCU is working fine because I can write custom code that host's a working access point using the Arduino IDE

karawin, 2017-10-18:
Flash it in DIO mode.

mailmartinviljoen, 2017-10-18:
Mr Karawin.

You are the man! it worked

martin-ger, 2017-10-20:
Thanks for the support!

closed 2017-10-20 by martin-ger