martin-ger/esp_wifi_repeater #87: Trickey Serial Console

mailmartinviljoen, 2017-10-19:
Hi.
First off this is an awesome project. Love it!

Setting up the Wifi extender via the console is a bit tricky. Its monitoring and outputting activity to the console all the time not giving you a chance to send a command unless you are quick. I had to press enter then quickly paste the setup commands pre-constructed in notepad.

It would be great if you could add a command that will allow you to initiate the log. I.E showlog then only will it output activity with a send any key to make it stop.

martin-ger, 2017-10-19:
Yes, debug output can be confusing - however there is no need to be quick. Debug output will not go into the input buffer. Just keep on typing the command and ignore, if it is scrambled - it will still work.

You might also give as a first command: "set auto_connect 0". Then it will stop trying to complain about a non-existent AP. However, you should reanable that, once the setup is completed otherwise it will try to connect only once after bootup and loose connectivity when the connection is lost once.

martin-ger, 2017-10-19:
Try it now: auto_connect (and thus the debug output) if off until you set your own ssid.

closed 2017-10-19 by martin-ger