martin-ger/esp_wifi_repeater #88: Problem to connect to a network when the signal is weak

destroyedlolo, 2017-10-28:
Hello,
I flashed my ESP-01 with current image of esp_wifi_repeater
 as well as ESP8266_NONOS_SDK one.

It's working perfectly well when the station signal is strong, but as soon as I go away and the signal weakens, the ESP become unresponsive. The blue LED flash 1 tenth of second and ... nothing after, no more more flash, no AP network.

I have don't have a console connected (waiting for my FDTI adapter to arrive), but or it is crashing, or it doesn't wait for the network.

Any tips ?

Thanks

Laurent

st0ff3r, 2017-10-28:
so I guess it does not restore connectivity when you move it closer to the station?

destroyedlolo, 2017-10-28:
No, it isn't restored.
But I did another test : I started the ESP with network coverage and then move to the place where it can't establish the connection.
And it is still working.
So it seems to me there is something bad during station establishment. But after, it can manage weak signal without issue.

martin-ger, 2017-10-29:
Hard to say without log of the serial output. On an ESP-01 especially the power supply might be a problem when the ESP becomes unstable. A bigger capacitor in parallel may help.

destroyedlolo, 2017-10-29:
I've added a 2200 uF and it seems it's working fine (more tests to do).
It seems initial connection to the network requires huge power and created issue.

Juste a small request : is it possible to publish an MQTT message as soon as the network is connected ?

destroyedlolo, 2017-10-29:
I'm replying to myself in case someone is having the same need : enabling the IP publishing (0x0002) is sending a topic as soon as the network is up and running on the ESP.

martin-ger, 2017-10-29:
That's right: "set mqtt_mask 0002" enables exacty that.

closed 2017-10-30 by martin-ger

destroyedlolo, 2017-11-01:
Hum, my ESP crashed also when the shutters go down (probably due parasites).
Anyway, the problem is deeper than that : I'm planning to power this WiFi relay using solar cells ... but as it seems the ESP doesn't restart automatically after crashes due to weak power, it's useless.
As soon as I got my FDTI adapter, I'll have a look on what's happened on the console and pursuit the investigations.

Bye.