martin-ger/esp_wifi_repeater #89: sleep command has some error over time

gustavovelascoh, 2017-10-31:
I was testing sleep command in the form 'sleep [secs]' and found that the module wakes up before the time expected. This is what I got:

for sleep 1800 (30 minutes), it woke up after 27 minutes
for sleep 3600 (60 minutes), it woke up after 52 minutes
for sleep 10800 (3 hours), it woke up after 2 and a half hour, approximately

Is this error expected or what could be the cause of this?

martin-ger, 2017-11-01:
This is probably due to the inaccurcy of the internal clock. Don't think it is a special of my software.

gustavovelascoh, 2017-11-01:
You are right. I've found that system_deep_sleep() function has a limit of 2^31-1 microseconds. That is approximately 71 minutes, so it is not possible to get ESP in deep sleep more that this time.

closed 2017-11-01 by gustavovelascoh