martin-ger/esp_wifi_repeater #92: Request Figure...  on AP mode when client connect ..

TridentTD, 2017-11-04:
Request figure :
Wish the esp-repeater's DNS  can translate 
the specified name to the esp-repeater's ap_ip.

such  as, 
esp.repeater.net   -->  192.168.4.1

If there is this figure ,  when users  connect  to  the esp-repeater on AP mode  
user will can use web-config  by  input url   by  http://esp.repeater.net
instead of   http://192.168.4.1

martin-ger, 2017-11-04:
This requires a DNS server on the ESP, something I tried to avoid until now. mDNS would be much simpler (tried this in my MQTT broker), but many clients-OS do not understand or use it.

TridentTD, 2017-11-05:
Thank you , master martin.

I have some questions to ask.  
In esp-open-lwip's dns   , can use for this figure ?
Or must requires another dns server ?
And what's difference between  DHCP's dns ?

mDNS , test by Android-OS 's chrome , I have ever tried and found that 
Android's chrome doesn't know the domain name from mDNS.
however, i will try the mqtt broker .  thank you.