martin-ger/esp_wifi_repeater #98: About auto_connect

MrMauro, 2017-11-19:
Hi!

After the changes related to issues #47 and #87 I understand that auto connect should be issued (if required) as no longer it is turned on by default.

May it be pertinent to include it on the Basic commands of the documentation to be aware of that?
Also this setting could be a candidate to be included in the Basic Web Interface so it is not required to enter the console when performing a basic setup.

Thank you!

martin-ger, 2017-11-19:
auto_connect is off (auto_connect 0) after first flashing or after "reset factory". As soon as you enter an SSID (on the CLI or the web interface) it will be automatically set on (auto_connect 1). As you usually typically set the SSID as first interaction, changing the auto_connect mode manually isn't really required.

There is no good reason, why you shouldn't want it to be on during operation, besides it's failure messages spam the serial console during first config. It doesn't harm anything on the web interface.

If you really want it to be off, you have to use the CLI. I think, that is okay, isn't it?

MrMauro, 2017-11-19:
You are right: auto_connect 1 is the expected in normal circumstances, and the way you describe its logic makes sense.
My confusion then may be related to a different cause during the provisioning of a device when its clients reported a drop on the connection that wasn't being recovered even after repeated restarts. Issuing the auto_connect resolved it, but I can't tell whether I have changed it or not as I did a lot of changes on the whole system during the setup.
I'll keep it in mind. However may you consider to include this explanation you just provided me about the auto_connect behaviour on the documentation? I think it's the only parameter missing there.
Thank you! The quality of your software is amazing.

martin-ger, 2017-11-19:
Thank you - added that to the docu in Readme.md.

closed 2017-11-19 by martin-ger