martin-ger/esp_wifi_repeater #103: NTP not working though NAT

adcomrod, 2017-12-17:
Hi, not sure if this is an issue or is it expected.
While testing the functionalities of the repeater I noted that ntp requests appears not to pass throgh the nat process. Tested an esp8266 as a client (running Espeasy) and the network time could not be obtained through the repeater (tried several servers including a local RPI acting as server on the network in which the repeater. Same for a laptop connected to the repeater. As soon as I got connected to the STA directly, NTP synchronices right away.
Regards
Adrian

martin-ger, 2017-12-18:
Certainly, it is not intended to be like this. In fact, NTP (like other UDP-based protocols) works (at least with other cleints like my esp_mqtt)...

I looked into the network trace and what happens is, that the esp_wifi_repeater in this case answers with "ICMP Destination unreachable (Port unreachable)". Strange that it does do this here and not for other requests. The only thing that might be special with this request is, that source and destination port are the same (both 123). Shouldn't be an issue, but I will further dig into that...

martin-ger, 2017-12-18:
Okay, think I found it: Line 828 in esp-open-lwip/lwip/core/ipv4/ip.c
Only registered source ports (>= 1024) are automatically mapped. Well-known ports (<1024) must be mapped manually. Could be changed there... Are you using the full build environment or just the precompiled firmware?

adcomrod, 2017-12-18:
Hi Martin. Glad I can help with this.
I am just using the precompiled firmware, since I am get use to work with Arduino IDE and its not available. Not sure when do you plan to release a new compiled update or how can I compile it. If there are a guide or something to compilemit, le me know.
Adrian

martin-ger, 2017-12-18:
Build a test version. Please try: https://www.cs.hs-rm.de/~gergelei/0x00000.bin and https://www.cs.hs-rm.de/~gergelei/0x10000.bin

adcomrod, 2017-12-18:
Perfect. As soon as I loaded it, it worked ok. As well for an esp and a laptop!!
Looks like is fixed.
Thanks!.
Adrian

sfilinto, 2018-04-11:
I just flashed using the pre-build .bin files and can confirm NTP wont work on my IoT devices ( ESP8266 )

martin-ger, 2018-04-11:
Okay, think, Ii know the fix... later today.

martin-ger, 2018-04-11:
Should work now...

martin-ger, 2018-04-18:
Don't think, that this has ever worked for ESPs. The problem was that the NAT code refused to rewrite requests FROM source ports < 1024 (privileged ports). In most cases this doesn't make sense.

But the ESP's SNTP sends its request from port 123 (I think, at least a privileged port) to port 123. This is not required from RFC4330 ("The Source Port field of these messages can be any nonzero value chosen for identification or multiplexing purposes."), but that's how the implementation currently is.

Now the ESP rewrites them to high ports...

imayoda, 2018-04-18:
thanks for explanation.. a long trip awaits :)

closed 2018-04-18 by martin-ger

imayoda, 2018-04-24:
hi there, I have a follow up on this.. 
I've been to the remote esp repeater and found out that incredibly it was passing NTP requests thru esp-repeater.. I investigated and found out it was the first pre-compiled automesh revision. :drum:  :smiley: 
Just for your information, kudos