martin-ger/esp_wifi_repeater #104: Feature request: Show STA MAC Address on Web-UI

projectb-temp, 2017-12-19:
Hi Martin,

first of all thank you for the great project you keep on going (and improving) here!

When using it on a simple ESP-01 with my FritzBox I came across a thing that might be worth adding:
I have MAC address based filtering for Wifi clients activated in my Router, so only known/whitelisted devices can connect via Wifi.

__How much effort would it be to add the MAC address of the used hardware (STA) as an informational text in the Basic Web Config Interface, so users can more easily copy&paste them into the routers MAC filter table (without the need of digging into telnet)?__

I think getting the MAC with something like __wifi_get_macaddr(STATION_IF, uint8 *macaddr)__ should be possible if I get the SDK right. But sadly I don't really know how to add it myself into the code and the HTML.

Thank you a lot for your feedback.

Best regards.

zawmintun1, 2018-01-02:
Yes should be have Basic Web Config Interface to easy.

projectb-temp, 2018-02-02:
@martin-ger 
Dear Martin,
as I got no response from you I'm pushing this request once.

Could you please be so kind as to look over my issue.
Thank you and best regards.

martin-ger, 2018-02-02:
Getting the MAC address is easy. The problem is, that the current web config is fairly simple stuff: it relies on the fact, that the page can be sent out in just one TCP packet. If I add more content to that page, I need a better sent_cb handler that continues the transmission. No rocket science, but for me more effort than simply looking into the CLI...