martin-ger/esp_wifi_repeater #105: NodeMCU slow connection

knutella, 2017-12-20:
I'm using a NodeMCU (Amica) and flashed the latest binaries (did all that DIO mode, clear the bin file stuff).  I am able to connect and am using Open authentication for now.  My download speed is incredibly slow (.08 mbps) and upload isn't too bad (.5 MBPS).  Tests done via Ookla speedtest and even local network connections are incredibly slow.  I often use these NodeMCUs around the house as MQTT sensors and have had no connection issues.  I'm next to the router testing with my android phone as the only client.

martin-ger, 2017-12-20:
If it works in general, flashing was surely correct. Guess the speedtest just overruns the buffer capacity of the ESP. Do you have the serial connected? Any output there? Does "normal" browsing work better?

knutella, 2017-12-21:
Normal webbrowsing to like weather.gov is next to impossible.  I may get like half a webpage after a few minutes of trying

from Serial:
add 1
aid 1
station: cc:61:e5:8f:67:a7 join, AID = 1
station: cc:61:e5:8f:67:a7 join, AID = 1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1

[image]
[image]

martin-ger, 2017-12-21:
LmacRxBlk:1 means: no more buffers. Happens, when the ESP is overloaded. It then drops packets and this leads to low TCP transfer rates. Happens sometimes, no real bug, but obviously your setup suffers much more from this behavior than others...

Which browser are you using?

knutella, 2017-12-22:
Android Chrome.

martin-ger, 2017-12-22:
Android (at least my Firefox) over WiFi sends a lot of layer 2 packets to the AP before starting the real web-traffic. Somewhere I read some (performance) rational for this voodoo. At least in my case, it does only consume bandwidth and buffers. Maybe this is a reason? Same behavior with a Windows browser?

Other possible investigations: what about the RSSI of your AP when you do a scan? If you have a Linux machine you might also make a trace of the actual packets...

knutella, 2017-12-22:
Tried with my busy windows 10 desktop, connects but says "no internet".  The signal is good but its channel 6 same as my router, probably wouldn't hurt to change channel.
[image]

marek26340, 2017-12-26:
I think you can't get rid of this issue because of the size of RAM ESP8266 has. I have a similiar problem, that is issue #40. 

BTW I'm getting 4 Mbit download and 2 Mbit upload, idk why is your speed so slow.

knutella, 2017-12-29:
I wonder if a ESP32 version would work better, I did buy a new router (not for this issue) and got the same results. .02 Mps download and .16 upload

martin-ger, 2017-12-29:
Don't think it is the processing speed of the CPU. Try "set speed 160", but I am pretty sure this will not change anything. Something is odd in your setting, as other do not experience this bottleneck.

Either your HF-situation is odd or some strange dependency between your devices and perhaps the software.

knutella, 2017-12-29:
Great news, I got 3 meg down / 4.6 meg up on my tablet Samsung Galaxy Tab E.  I guess my Moto G4 Play has some oddity with this device.  I'll see if there are any settings on it to change.  it otherwise has great speed on my normal router.

closed 2017-12-31 by martin-ger