martin-ger/esp_wifi_repeater #110: Is possible AP NOT connected to with Root Ap internet ?

tinamore, 2017-12-28:
I have a house 4 floors. Internet Router on 1st floor, and 3 esp_wifi_repeater (Automesh) located on 2nd floor, 3rd floor, 4th floor.

I'm wondering is possible  (Esp floor 2) 
 connected to (Esp floor 3) and (Esp floor 3) connected to (Esp floor 3). Apparently there is not any ESP connected to the Internet Router
 => Can not internet in mesh esp_wifi_repeater.

What did I miss?

martin-ger, 2017-12-28:
Should not happen: In automesh mode the ESPs are first looking for an AP that has NOT 24:24 as prefix Try to start the one in floor 2 first - it should connect to the router. Then start the other's

The isea is, that it works in any order, but just to make sure...

tinamore, 2017-12-28:
Great ideal,
This is a very great project about esp8266.

tinamore, 2017-12-28:
I have a building 5 floor and each floor has 7 room and each room has 1 esp8266 will send the temperature and humidity send every 5 second (use httpclient), this esp will connect to mesh esp_wifi_repeater.
WiFi router placed at the ground floor.
I think the httpclient of esp8266 of each room will send small data and mesh esp_wifi_repeater would respond well.
---
I can do this ?

offtherails2010, 2017-12-28:
howdy folks !

Just stumbled upon this project and just HAD TO SAY - WELL DONE MARTIN & TINA !

This is EXACTLY WHAT I NEEDED !!!

Low cost open source solution for a NAT Router Repeater - you folks have saved my BACON ! lol

Also, as a quick question for the last post made;

Would it also be possible to send all this data per room for the temperature/humidity (is it a dual temp & humidity sensor per ESP8266 ??) to a hand-held device that doesnt need computer ???

Like a Nokia LCD Screen that gives all readouts ?

Possible update on the project ?

i'd be willing to help as much as i can, i make my own PCB's for years now, so if you need help with PCB Design, please do let me know, i would be very honoured and pleased to help & be involved in this project

Sadly i only use Cadsoft Eagle for PCB Designing, as long as you can give me a schematic that is correct, i will be able to design the PCB for you, and custom doesnt mean expensive, it is so damn cheap to get PCB's made these days, like for example;

A 50x50mmm sizes PCB, 1.6mm thickness, 1 oz copper, times 5 pieces, would only cost about £15 USD to manufacture professionally

So please do see if we can make a little hand-held device to give readouts, so we can walk about freely and check the temp/humidity in any room in the house - from anywhere in the house !

Naturally it would have to be Lipo battery powered, which would also be very awesome to make, i have many Lipo charging PCB's that have USB Mini-B or USB-Micro-B connectors on them, a few slim Lio batteries meant for Tablets, so powering the hand held device would be really cool

**# But i am USELESS AT CODING -** i cannot code to save my life lol
So i am here to beg you guys to see if you can do this, or if it can be done, i dont know, if it can it would be really shweet :)

If not, no worries, i will still offer to help with any PCB designing, regardless :) and thanks so much in advance :)

tinamore, 2017-12-28:
@offtherails2010 Thanks for infomation.
But i want to send of temp & humidity from Earch room via httpclient of esp8266. All data will save to database in Cloud (Azure or AWS).

offtherails2010, 2017-12-28:
most welcome Tina & no worries i understand :)

@martin-ger - would my question be possible or should i open a new thread for my question so as not to mess up Tina's help requests ?
thanks loads in advance :)

martin-ger, 2017-12-28:
5x7 HTTP-requests per second are quite a lot, especially if you think that each HTTP-request is a TCP connection, i.e. min 6 packets, and these packets are repeatet at least on average 2 times, i.e. about 630 packets per second. Also, the ESP doesn't allow for more than 8 connected WiFi STAs.

Can work, but I would consider two things: Using MQTT or CoAP as transport protocols (about 2 packets per transaction) and reducing the interval (temp and humidity aren't really changing in the range of seconds - if so, something is going wrong really badly ;-) )

martin-ger, 2017-12-28:
@offtherails2010: perhaps another thread. Thanks for the offer. To read the values from a mobile tablet, I would suggest to put a central RasPi in the middel that runs am MQTT-Broker, perhaps NodeRed and a mobile device that gets the values or even a prepared web-page from there.

tinamore, 2017-12-28:
@martin-ger 8 "ESP doesn't allow for more than 8 connected WiFi STAs". I can use 2 esp_wifi_repeater for one floor. Total: 10 esp_wifi_repeater (Can max 16 station).

According to you then 10 esp_wifi_repeater can cause interference wifi channel beacause mesh mode use same channel wifi?

Thanks for response.

offtherails2010, 2017-12-28:
@martin-ger &
@tinamore 
my humble apologies for the disturbance on this thread, i will create i new one with my question and the help offer is always available for you both, anytime, anywhere :)

& thanks martin, i will see if i can grab a raspi, i think the cheapest i could find was on aliexpress and its called an orange pi, naturally id choose the smallest package size available lol

Also that would be my first raspi id ever own so id be very NOOB worried about how to program it correctly haha, more Q;s in another thread and my humble apologies again :)

martin-ger, 2017-12-28:
You can use 2 ESPs per floor and you probably need them anyway to cover all rooms. However, normally you do not enforce a special client to connect to a special AP, they simply choose the one with the highest RSSI. So balancing the network isn't so easy, but as the nodes are static - can work.

Guess, interference shouldn't be a big issue. To be honest: I have no experience with a network of 10 repeaters or more and over 30 clients.

tinamore, 2017-12-28:
@offtherails2010 
I used orange pi and other computer pi from China. Poor quality,  running for several days is dead.

Raspberry can run a year and still very good.

martin-ger, 2017-12-28:
@offtherails2010: I would go for an original RasiPi. I also bought an OrangePi and had a lot of trouble with it. Ended up with an old RasPi B as home server. Especially, if you are a NOOB, the real RasPi is much easier to master, as you can use the standard Raspian Linux distribution and all install instructions work right out of the box.

Saving 20 Bucks isn't worth all the frustration you will experience with a non-standard HW...

tinamore, 2017-12-28:
Thanks for your information. I am having 20 esp8266 and i will test with esp_wifi_repeater.
@martin-ger I really feel lucky to meet your project. It helps me to solve the problem. You are an excellent programmer.

offtherails2010, 2017-12-28:
howdy guys :)

thanks loads for the advice, i will try to find a RasPi at lowest cost possible as im on NO income at the moment lol so projects are kinda all on hold for the mo except a very select few, this being one of them as my mother always complains about there being weak wifi signal downstairs, so a repeater is welcomed lol

i bought these two to make the repeater using an **[ESP07](https://www.aliexpress.com/item/Smart-Electronics-10PCS-LOT-ESP-07S-ESP-07-Updated-version-ESP8266-serial-WIFI-model-Authenticity-Guaranteed/32776272948.html)** & a **[wemos](https://www.aliexpress.com/item/CH340G-USB-TTL-converter-for-ESP8266-ESP07-ESP12-to-WeMos-D1-mini-with-pin-headers-set/32836543446.html)** compatible board

Only reason why i didnt go for a pre-soldered ESP/Wemos board is so i could add my own antenna to the ESP07 via pigtail to SMA male connector which will be screwed in place in a small enclosure, then an antenna/aerial screwed into the SMA Connector :)

closed 2017-12-30 by tinamore