martin-ger/esp_wifi_repeater #111: Anyone else having 'Access is denied' trying to flash w/ ESP Tool on Win 10?

jbishop129, 2017-12-28:
Apologize in advance for this, I realize it is sort of off-topic to the core of the project. I use the ESP FLash Download tool (latest: v3.6.2.2) in Win 10. Tried running the .exe normally, as well as 'Run as Administrator' with same result: I select the two .bin files, correct COM port, baud, click Start, and get 'Com Fail'. In the background cmddline screen I see ```[2017-12-28 09:24:11,217][ESP8266Loader_spi[1]][espDownloader.pyo][line:436][ERROR]: could not open port u'COM5': WindowsError(5, 'Access is denied.')```

Can't figure out why it wont talk to the device. Running a generic ESP8266 @4MB flash (I set the Flash Size in the tool to '32Mbit'). Probably something dumb and easy, any pointers?

jbishop129, 2017-12-28:
I'm an idiot. Disregard.

closed 2017-12-28 by jbishop129

nicholas-gh, 2017-12-28:
Please do still post the answer, since you posted the question :) 

https://xkcd.com/979/

jbishop129, 2017-12-28:
Lol, fair enough: 
I had been flashing several ESP8266 chips today through the Arduino IDE, and I "thought" when I had the above problem, I had verified I still had the ability to communicate with said chips via the IDE, but I must have had my order of events out of whack. I was unable to talk to the ESP chips via the IDE either. A simple reboot of my Windows 10 PC resolved the issue, I was able to flash martin-ger's firmware on several ESP8266's using the "ESP8266 Download Tool" after the reboot.

That said I'm enjoying playing with the Mesh Mode with three ESP8266's! Certainly not fast, but does work pretty dang well! cheers @martin-ger for this code!