martin-ger/esp_wifi_repeater #113: ESP NAT Router Hand-Held battery Operated Temp/Humidity Readout Device Possible ? Also ESP32 Question

offtherails2010, 2017-12-28:
@martin-ger 

Firstly apologies for hijacking the previous thread with my Q's and pestering lol

Would it be possible to send Temperature & Humidity data per room in my house and one for outdoors to also check the Temp/Humidity so i dont even have to open a window to see how hot/warm/cold/freezing it is lol

So this house has a loft, 3 rooms upstairs and 3 rooms downstairs, excluding the Bathroom, would it be possible to send all this data to a hand-held device that doesnt need computer ???

Like a Nokia LCD Screen that gives all readouts per room all on-screen ?
So thats 8 ESP devices each with a Temp/Humidity Sensor, for my 6 rooms (3 per floor) plus 1 in the loft & 1 outside...

Possible update on the project ?

Maybe even using an ESP32 as the hand held device's brain ?

i'd be willing to help as much as i can, i make my own PCB's for years now, so if you need help with PCB Design, please do let me know, i would be very honoured and pleased to help & be involved in this project

Sadly i only use **Cadsoft Eagle** for PCB Designing, as long as you can give me a schematic that is correct, i will be able to design the PCB for you, and custom doesnt mean expensive, it is so damn cheap to get PCB's made these days, like for example;

A 50x50mmm sizes PCB, 1.6mm thickness, 1 oz copper, times 5 pieces, would only cost about £15 USD to manufacture professionally **(Even Less Money if we all contribute, those who are interested that is, if no one interested then i's definitely still be going through with it lol)**

So please do see if we can make a little hand-held device to give readouts, so we can walk about freely and check the temp/humidity in any room in the house **- from anywhere in the house !**

Naturally it would have to be Lipo battery powered, which would also be very awesome to make, i have many Lipo charging PCB's that have USB Mini-B or USB-Micro-B connectors on them, a few slim Lio batteries meant for Tablets, so powering the hand held device would be really cool

**# But i am USELESS AT CODING** - i cannot code to save my life lol
So i am here to beg you guys to see if you can do this, or if it can be done, i dont know, if it can it would be really shweet :)

**# ESP32 Question:**
Lastly, would your NAT Router firmware be compatible with the **[ESP32](http://espressif.com/en/products/hardware/esp32/overview)**, to make it run faster/better with clock speed up to 240Mhz

Plus it has Bluetooth too so dont really know what to do with that unless youve got some cool idea's lol

If not, no worries, i will still offer to help with any PCB designing, regardless :) and thanks so much in advance :)

RudyFiero, 2017-12-29:
And because I read all of the above, I'll throw this up for the hell of it. My attempt at a small WiFi remote. Hacked together with available parts. This is the first one I made. I have been thinking about doing a PCB to make assembly easier. 

I was going to do a remote box with more switches and larger. But I felt that it should have a display. But I had none that fit the enclosure I wanted to use. This oled would fit but it would have looked silly since it didn't fill the space available. But I changed my approach when I decided to use the oled for a display. But I was going to make it as small as I could while still using available components. 

I took these pictures before I put in a vibration motor. There still is room for something else below the ESP module but I have not thought of what else to include. Not obvious is the orientation sensor in the corner of the LCD. I used an ESP-07 because it was a bit smaller and I thought I needed the space for the voltage regulator. 

The power switch was included because I wasn't sure how long the the battery would last when it was in sleep. If the slide switch is left on the unit will power off after a timeout. The push buttons on the front can wake the circuit. Currently I have two separate modes of operation depending on which switch woke the device.

[image]

offtherails2010, 2017-12-29:
Hot damn wow lol nice job man ! yeap it shows youre skilled at electronics design!

Great job getting that into the tiny project box !! and your cut-outs look perfect !!

i think thats the size project box i have for the Wireless NAT Router Repeater, just about to dig out my soldering iron any day now, almost got to it lol

Would you consider having the PCB open source by any chance ?

martin-ger, 2017-12-29:
As said in the other thread:  To read the values from a mobile device, I would suggest an architecture with a central MQTT broker. 

Basically, you will need four types of nodes:
- Temp and humidity sensors. There are plenty of projects around, that implements these based on MQTT and the Arduino environment. Here the challenge is power supply: do you have permanent power or do you want to run on batteries. If so, clever power management (i.e. deep sleeps) is required.

- esp_wifi_repeaters: to extend the WiFi range (if required). Can be used as is. They could be extended to host the sensor software as well, but this is significant app-specific coding. These nodes require permanent power supply (need 70mA all the time).

- the central node: a RasPi in the middel that runs am MQTT-Broker, perhaps NodeRed. One could think of using an ESP8266 here as well with my MQTT broker software (https://github.com/martin-ger/esp_mqtt )
  
- a mobile device that gets the values or even a prepared web-page from there. Either a simple tablet or old smartphone (can get it for about 50-100$) with just a browser or an MQTT client. Or a pure custom device like the one from RudyFiero (really nice platform!!) with handcrafted new SW, but this is surely lots of work.

My SW currently does NOT run on ESP32. There is no obvious reason, why it cannot be ported to it, but the API on the ESWP32's FreeRTOS SDK and the ESP8266 NONOS SDK are different, so it definitly would need some work. 

You increase CPU speed even on the ESP8266 to 160 MHz (set speed 160 in my CLI), but this doesn't speed up the WiFi repeating significantly, as buffers and bandwith are the main issue.

RudyFiero, 2017-12-29:
@offtherails2010  I would probably make the gerber files available if someone wanted them. I use a pcb software we use at work and it isn't compatible with the ones hobbyists use. I did a quick look for the plastic box on AliExpress and I have not been able to find it. No surprise. Part of the appeal for this project was the small size and the way it all fit. I don't have a lot of time to make a pcb at this time but I plan to this next year. Better to find other options. 

I agree with Martin in the above post. I have gone the Raspberry Pi - MQTT approach. I have done a little bit with the ESP32. While the ESP32 runs at 240Mhz that isn't so much faster than the ESP8266 at 160Mhz. While there are more options available with the ESP32 due to more hardware, I think the ESP8266 does most small IO control applications.

offtherails2010, 2017-12-29:
howdy guys !
Firstly, a major thanks for the info and responses, im in absolute awe !!!

Also thank you so much for the great advice;

@martin-ger - thanks - i will address those 4 issues individually, now that i have nothing but time on my hands, unemployed and cant walk through disability, so iim in bed 23 hrs a day atm..
this sounds like a great start and i will be using battery powered devices all around the house and especially outside

But if any internal temp/humidity sensors can be easily connected to a wall wort 5v adaptor then thatll be easier, i do have a few 18650's ive been saving for the tast, one/two per device or do as Rudy has done and go for the smaller Lipo flat-batteries - to which ive also set aside some old Sony Ericsson BST-38 flat lipo batteries and have an assortment of other smaller ones too.

So theyll definitely have the charging PCB in them, also have some of these

@RudyFiero ;
That little project box you got everything into just looks stunning, man !!

Thats pretty much EXACTLY the sort of hand-held device im looking to create so one can walk about freely and without a computer, wirelessly and check different room's temps and the outside temp/humidity too

Even if i have to use a RasPi, as martin has mentioned to me in the last few posts that its just less headache to get an original RasPi instead of the smaller OrangePi's fromm China, so a RasPi B or something would be great but ive never used it yet..

As for the PCB - i use cadsoft EAGLE - If you would be kind enough to furnish me with a schematic - i'd be able to create a PCB from it and upload it to my github where you'd haave full access to it for modifications/alteration etc

What do you think about that Rudy ? have you used Cadsoft Eagle before ? its free to use upto a certain PCB size, which ive found way-more-than-enough for my needs from the last few years

i dont have much to do so im pretty free, in a lot of pain hourly so my time im able to work on these things is limited to a few hours during the daytimes, but other than that im pretty free to come up with a PCB for the tiny lil cut and gorgeous device youve made here, plus a PCB for an ESP-07 temp/humidity sensor board.

i think this would be an awesome collaboration between us if its okay with you, plus however youd like to run the licence on these designs will be up to you, i lean more to completely open source but as its your design, its your call :)

id just be after an attribution/share-alike kinda licence :)

i also have a very old set of PCB's that use the Arduino NRF24L01 combinations for this exact thing plus a Nokia-LCD PCB for wireless readouts, but being RF it really is limited....

Being wifi and using a few well placed Repeaters id have a better signal thru walls and doors etc than with the RF, as in the furthest 2 parts of the house, one of the upstairs rooms and the end of the garden, RF is extremely weak and doesnt even work sometimes, but my wifi still have a good signal at these 2 points

By the way, thats the EXACT project box i have for the repeater @RudyFiero !! nice find on aliexpress, thats a really great price !

Anyhow, what do you think guys ?

offtherails2010, 2017-12-29:
EDIT;

i will probably use the ESP-07 as its really tiny and GIVE's me the option of using a pigtail i.FL to SMA Antenna so that i can connect up a proper aerial to it for better signal

Here is the **[ESP-07](https://www.aliexpress.com/item/Smart-Electronics-10PCS-LOT-ESP-07S-ESP-07-Updated-version-ESP8266-serial-WIFI-model-Authenticity-Guaranteed/32776272948.html)** and

Here is the **[Pigtail](https://www.aliexpress.com/item/Free-shipping-2pieces-lot-Mini-PCI-U-FL-to-SMA-Antenna-WiFi-Pigtail-Cable-IPX-to/32310278154.html)** im waffling on about lol

offtherails2010, 2017-12-29:
Some of my PCB's;
**[Low-Battery Gauge](https://imgur.com/jXer4j3)
[Low Battery Gauge Pic 2](https://imgur.com/6Wk4Mnw)

[Nokia LCD Backboard PCB front](https://imgur.com/GmhYMcs)
[Nokia PCB Back](https://imgur.com/I8vcID8)

[Arduino SLIM SBBB](http://i1246.photobucket.com/albums/gg616/offtherails2010/IMG_0398.jpg)** (SBBB = Super Bare Bones Board !) This PCB ive put in, Under-IC LED Lighting, LOL, i do have all colours, but i stuck with a Blue LED one for a long time now lol this is my own designs PCB's - all made from Eagle, Proofed home-brew PCB, then once bugs removed if any, it gets manufactured, small batches :)

RudyFiero, 2017-12-30:
@offtherails2010 The software package I use is called Pads and a couple of years ago it was priced at $5000 for the basic platform. I would not be interested in learning a new package. 

I don't have a schematic of the circuit. Since I didn't make a PCB I only had a few partial pencil sketches. At this point I will mention that this is not an appropriate place to discuss this. It isn't directly connected to this fantastic repeater program. When I have anything more that I can post on my project I will do it on http://www.esp8266.com/viewtopic.php?f=11&t=15929

offtherails2010, 2017-12-30:
@RudyFiero 

WOW £5K !!! i think Eagle is kinda like that too, very high priced if you9r running a business i guess its affordable, kinda..

Apologies in advance Rudy, but this is why i made this forum post so as not to hijack the other one with my idea, and its still concerning the NAT Router Repeaters because i will need more than 1 for the whole house, i think im looking at 3, i have the parts to make 2 of them, but not a third, will see how i get on and flaqshing the new firmware, kinda excited now lol !

So this post is for

>> 4 issues that martin pointed out in the last post

>> 4/5 ESP-07 Temperature/Humidity Sensors, most battery powered via lipo

>> 3 Repeaters to bathe the whole house in lovely wireless internet lol

As youve made your new post for specifically for ESP with OLED & Lipo, so we can jump in there and leave the other thread thast we were in clean and back to the point of its main title.

I know ive been out of touch for many years from all internet, but i do try to keep the point going on the very subject ive asked for or im into, sometimes i fail so i do apologise, but i dont think we were in much of a violartion from the T&C's, a moderator will come along and tidy stuff up if needed :)

Anyhow, enough of that and back to the point lol

Ive been working on a Nokia LCD Readout device, a hand held device so that it detects the temp/humidity sensors while being boosted by the repeaters, but this was for another project, a watercooling project that actually CHILLS the water lol

So i think the hardest part of this for me will be the hand held device firmware coding.. ugh, totally NOOB useless with that...

tinamore, 2017-12-30:
@offtherails2010 
You try use raw packet wifi (No need for wifi connection between AP vs Station)
The client sends the raw packet (id of client and value store mac address) and master can sniff packet.
I tested 1 master esp8266 in 1 second can receive enough value of 30 clients esp.
```
Send raw packet: https://github.com/markszabo/FakeBeaconESP8266/blob/master/FakeBeaconESP8266/FakeBeaconESP8266.ino
```
```
Sniff packet:
https://github.com/kalanda/esp8266-sniffer
```
with 1 Nokia LCD master can read value 30 devices.
Hope my information can help you.

offtherails2010, 2017-12-30:
@tinamore 

thanks for the advice, it indeed helps, ever bit of advice helps no matter how small, plus i appreciate it & am very grateful for the help :)

ive dug out the Peltier Watercooling PCB with Nokia LCD, so im just cleaning it up for porting over to this new Wifi Repeater/Temp/humidity sensors, once again, thank you Tina, you folks are truly awesome !!

offtherails2010, 2017-12-30:
@tinamore 

ahh if that isnt this repeater firmware projects, i wouldnt be able to do it, thanks anyways though, i do need the repeater hardware & software though :)

offtherails2010, 2017-12-30:
Ugh accidentally closed this post, reopened again..

martin-ger, 2017-12-30:
For the repeater any ESP8266 board will be fine: a Sonoff Basic is probably the cheapest option for an ESP with mains power (around 5$) even if you don't need the ralais.