martin-ger/esp_wifi_repeater #114: Firmware compiled with SDK 1.5

jadolg, 2017-12-29:
I was using firmware compiled with this SDK version and it suddenly disappeared from the repository. Can you please put it back or show how to use the other one? I'm using esptool.py to flash and when I try to use the one here I get a lot of weird characters when I do `screen /dev/ttyUSB0 115200`

martin-ger, 2017-12-29:
The reason it disappeared was that I found the problem wit the newer one on some ESPs:

Sometimes, especially on old ESP-01s, there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/espressif/ESP8266_NONOS_SDK/raw/master/bin/esp_init_data_default_v08.bin and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1).

Could you try that?

jpumford, 2018-01-01:
I had the flashing LED/garbage serial issue and this fixed it for me, thanks so much!

jadolg, 2018-01-01:
[image]
this is the module I have. I'll try flashing this but I have no idea which of those it is :-(

martin-ger, 2018-01-01:
This is a Wemos D1 Mini, use the settings for 4MB.

jadolg, 2018-01-04:
yes, that solved the issue. thanks for your help and for developing this project. it's awesome! :)

closed 2018-01-04 by jadolg