martin-ger/esp_wifi_repeater #116: How to erase code off NodeMCU

knutella, 2018-01-02:
I had an unexpected experience today, I re-purposed the NodeMCU and loaded a short Arduino sketch on it.  Lo and behold I noticed a few days later that the MyAP was still active on my network.  The NodeMCU was running both my Arduino LED Sketch and this code also.  I had loaded the code at 0x3fc000 if I recall and I guess it was able to run both codes at the same time.  I had to completely erase the chip based off this [link](http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/)

martin-ger, 2018-01-02:
Thta's normal. The ESP stores its WiFi Config at 7E000h. This is just the config of the STA and AP interface, not the complete program. If you reflash it with a program, that doesn:t need WiFi (like blink), it just starts with the last config. If your program does config the WiFi interface, it overwrites that.

RudyFiero, 2018-01-02:
It has nothing to do with the esp_wifi_repeater code. It is the underlying structure. If you don't disable it then it will be there to execute next boot.

closed 2018-01-02 by knutella