martin-ger/esp_wifi_repeater #117: Setting a new STA MAC address does not work

electronking, 2018-01-02:
Happy New Year!  I believe I may have discovered a bug in the firmware, regarding changing the STA MAC address.  First I telnet into the repeater and change the ap_mac and sta_mac address using the set command.  Next I save the configuration.  After power cycling the ESP8266 board, the repeater connects with my router but the Active DHCP leases on the router show the repeater is still using the original esp8266 MAC address.  The serial output from the ESP8266 also shows the original MAC address under "mode: sta(xx:xx:xx:xx:xx:xx)".  When I telnet into the repeater, and issue a SHOW CONFIG, the STA MAC is the one one that was configured with the SET command.  The AP MAC seems to work and is consistently reported the same in all cases (router DHCP leases, telnet show config, and serial output.

zawmintun1, 2018-01-02:
Happy new year !

Please told me how to change  AP & STA MAC address of ESP.

martin-ger, 2018-01-02:
@electronking : Think you tried to set a MAC address with an odd first byte, but the docu of wifi_set_macaddr says: 
> The bit 0 of the first byte of ESP8266 MAC address can not be 1. For example, MAC address can be “1a:XX:XX:XX:XX:XX”, but can not be “15:XX:XX:XX:XX:XX”.

With an even first byte of the MAC addr it works fine, with an odd it fails.

martin-ger, 2018-01-02:
set [ap_mac|sta_mac] xx:xx:xx:xx:xx:xx: sets the MAC address of the STA and SOFTAP to a user defined value

electronking, 2018-01-03:
@martin-ger:  Yup, you are correct, thought I had completely followed the detailed instructions you provide but somehow missed the info on bit 0.  Just a case of me not completely RTFM :)  Thank you again for this project!  I plan on using it to learn how to create programs for the ESP8266 using the esp-open-sdk platform since your project has so may useful functions.  May your New Year be filled with good health and many successes.

martin-ger, 2018-01-03:
Actually, I added this hint after your issue. ;-)
I was aware of it from the SDK docs, but I forgot to mention it for the users.
Thank and best wishes to you as well.

closed 2018-01-03 by martin-ger