martin-ger/esp_wifi_repeater #118: Error codes

HLNZ, 2018-01-03:
Hi,
I am having trouble with my units connecting to the wifi router. I am getting disconnection error results, but I'm not sure what they mean.
What do all the error result codes mean / where do I go to find the proper descriptions.

Thanks.

martin-ger, 2018-01-04:
These are generated by the SDK. Havn't found a description so far.

HLNZ, 2018-01-04:
Hi Martin
Can you tell me what error codes 2,3,4 and 204 are?
Or what error codes mean bad username, bad password , wrong network?
I think that I have the unit setup correct, but it is having a problem both
making its own ap and connecting to an ap.
I'm using an basic esp8266 module.

regards ,
Robert

HLNZ, 2018-01-04:
hi Martin,
one further question, exactly what tool chain are you using to compile the
code on git hub?
I looked at the link to the other git hub project that you mentioned,  but
I still haven't  figured it out exactly.

regards,
Robert

martin-ger, 2018-01-04:
I used the esp-open-sdk. Once you have installed it, you just have to adjust the path in the makefile, it compiles right out of the box.

HLNZ, 2018-02-08:
Hi Martin,
I am still stuck trying to get this to work. 
I am sure that the unit has accepted the latest firmware you posted.
I can talk to the esp, and I have a constant on red LED.

1) does the esp need to be connected to an AP to create its own AP?
2) the esp's AP does not show on either my PC or my phones wifi searches. how can I tell that the esp has correctly created an AP?

If you have any other troubleshooting tips, I'm all ears....

Regards,
Robert

martin-ger, 2018-02-08:
After flashing connect to the serial port with a terminal program (e.g. the Arduino serial monitor) and 115000 bps. Here you should see some status output, when booting correctly. If so you can enter any config command on the command line.

HLNZ, 2018-03-02:
Hi Martin.
I have found the issue with my setup, it was a (lack of) power supply issue.
The firmware is now running as it is supposed to.

Robert.

martin-ger, 2018-03-02:
Great news! :+1:

closed 2018-03-02 by martin-ger