martin-ger/esp_wifi_repeater #119: How to AP node Mesh limit 7 station client?

tinamore, 2018-01-05:
Hi,
I have the esp_wifi_repeater model with Mesh Mode as follows
```
Floor 3:  ESP_WIFI_REPEATER_3(with mesh)

Floor 2:  ESP_WIFI_REPEATER_2(with mesh) Station_1 Station_2 Station_3 Station_1 Station_4 Station_5 Station_6 Station_7 Station_8

Floor 1:  ESP_WIFI_REPEATER_1with (mesh)
```
When the building has electricity, Maybe 8 stations in the 2nd floor will connected to ESP_WIFI_REPEATER_2  (AP of ESP_WIFI_REPEATER_2  limit max connection 8 station)

ESP_WIFI_REPEATER_3 will not be able to connect to ESP_WIFI_REPEATER_2 (full 8 station). Floor 3 will not have internet.

How can ESP_WIFI_REPEATER_2 have up to 7 connections with mac id not match 24:24 .... will leave a seat for 24:24 (for ESP_WIFI_REPEATER_3 )?

marek26340, 2018-01-05:
I don't think that the esp_wifi_repeater was programmed to reserve one WiFi client slot for the mesh.

marek26340, 2018-01-05:
Oh and also I don't think that I fully understand your question(s).

tinamore, 2018-01-05:
@marek26340 Yes, You're right, my purpose isreserve one WiFi client slot for the mesh.

martin-ger, 2018-01-05:
The mesh connection needs one client slot, that's right. However, there are no reserved slots. First come, first serve.

closed 2018-01-09 by tinamore