martin-ger/esp_wifi_repeater #121: After reflash no  AP

Italque, 2018-01-05:
Hi after a week I decided to reflash my ESP8266. This time at the first boot there isn't a "MyAP". I tried many times to erase flash (also with esptool) end write it again. Every time there isn't an AP. Obviously the ESP8266 is not burned. What could I do?  Thanks in advance.

martin-ger, 2018-01-05:
Did you try DIO mode?

Italque, 2018-01-05:
Thanks for reply! Yes I tried DIO mode as said in the Readme.

Inviato da iPhone

martin-ger, 2018-01-05:
Pleased check the output on the serial port.

Italque, 2018-01-05:
Is there a way to check the serial output inside the Espressif flash download Tool?
Could you point me the right way to do this?

Italque, 2018-01-07:
I used putty for the serial port at different baud values but the result is not understandable. I also uploaded an example sketch for a Soft AP and it works.

martin-ger, 2018-01-07:
Did you try this: 
Sometimes, especially on old ESP-01s, there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/espressif/ESP8266_NONOS_SDK/raw/master/bin/esp_init_data_default_v08.bin and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1).

Italque, 2018-01-07:
Thanks for your help!
I had flashed the wrong sector with the "esp_init_data_default.bin", in my case. I have an ESP-12 but I had flashed the 0xfc000 sector although the 0x3fc000 was the right sector.
Now everything is solved!

Inviato da iPhone

closed 2018-01-11 by martin-ger