martin-ger/esp_wifi_repeater #123: Problems connecting to an open Wifi (no password)

PieBru, 2018-01-06:
Hi, I'm not able to connect the ESP router as STA to an open AP (no password).
Both the web ui and the telnet console doesn't accept a null password, and the value remains "password".
Any hint?
Thanks,
Piero

PieBru, 2018-01-06:
After some code digging I obtained a partial meshed solution by configuring my settings directly into user_config.h (SSIDs and passwords) and config_flash.c
```
    config->auto_connect = 1;
    config->automesh_mode = 1;
    config->phy_mode = 1;  // 1 = the good old B mode (it may be superfluous, but sometimes it helps)
```
Before flashing the firmware I cautelatively erase the whole SPI flash. This way I also clear the MAGIC_NUMBER so at the first boot the firmware initializes the settings:
```
$ esptool --port /dev/ttyUSB0 --baud 115200 erase_flash
```
The ESP router seems to work, but two problems remains:
1. The ESP router AP ssid isn't the same as the STA ssid, but it's ESP_XXXXXX (mac least significant bytes), which I did not configure. It smells as an ESP microcode takeover.
2. The ESP router http page works, but now it refuses telnet connections on port 7777.

It would be nice to be able to transparently mesh even open (no-auth) wifi networks.
Thanks,
Piero

marek26340, 2018-01-07:
You did all this for nothing. All you have to do is type pretty much any password shorter than 8 characters in set password command.
Example "set password nothing" or "set password weed" and etc.

Also you have messed up the config. You really don't need automesh mode turned on, unless you want to link more than 1 ESPs together.

Now I suggest you to reflash the ESP with the original code (the code that you did not modify), and everything should work now. But before that you should clean the entire flash memory with a 1MB blank.bin file and a .bin flasher. For a 4MB flash size, you should clean address 0x000000, 0x100000, 0x200000 and 0x300000. When I was working with a ESP flash tool from espressif like you, it created an AP like your's did, and that's wrong.

martin-ger, 2018-01-08:
"set password none" should do the job.
To reset all settings to default use "reset factory"

marek26340, 2018-01-09:
he changed the default settings in the code, that's what i'm talking about...

PieBru, 2018-01-09:
Thank you all, sorry for my late reply.

It is not clear from the doc that a password shorter than 8 chars leads to an empty password. For the newcomers as me, I also suggest to update the doc with a simple paragraph on "How to repeat an open AP".

My interest in this awesome project is motivated by: 
1) better coverage of my home wifi (encrypted) network without adding visible tech boxes all around, to support IoT as well as personal devices; a mesh solution will be ok with some kind of dynamic adaptation;
2) extend my guest open wifi network in my garden, 3 or 4 solar-powered mesh hops should be enough.

PieBru, 2018-01-11:
Tested, it works with stock firmware and "set password none".
IMO the doc should explain that any password shorter than 8 chars leads to an empty password.
Closed, Thanks.
Piero

closed 2018-01-11 by PieBru