martin-ger/esp_wifi_repeater #124: Feature request : Default to sleep power management mode

1rabbit, 2018-01-09:
Hi,

Excellent software you have there, I'm using it outdoors as an experimental mesh with low cost nodes https://imgur.com/a/jXTOc unfortunately the batteries don't allow me an unlimited run time.

It would be interesting to add an option that would :

If set to 0 = vanilla mode, doesn't change anything

If set above 0 = number of minutes to sleep before checking again for original AP ssid to connect to/repeat. If original AP is non existent/out of reach, going back to sleep.

That way it would be possible to deploy battery powered nodes that would only be using power when needed. You would put them to sleep by either disabling or removing the original AP.

Ideally something with more settings, that would behave like this :

sleep_after_X_minutes_without_ap = 30
seconds_between_retries = 120
number_of_retries = 2

On a timeline mm:ss

00:00 > Starts looking for AP
00:05 > no AP found, going to sleep for 120 seconds
02:05 > Starts looking for AP
02:10 > no AP found, going to sleep for 120 seconds
04:10 > Starts looking for AP
04:15 > no AP found, going to sleep for 30 minutes
34:15 > Starts looking for AP
34:20> no AP found, going to sleep for 120 seconds
36:20 > Starts looking for AP
36:25 > no AP found, going to sleep for 120 seconds
38:30 > Starts looking for AP
38:35 > no AP found, going to sleep for 120 seconds
40:35 > Starts looking for AP
40:40 > AP found > Repeat till AP disappears then go back to sleep mode.

Keep up the great work !

martin-ger, 2018-01-09:
Will think about that. Like the idea of a battery powered outdoor mesh.
One issue: you are using ESP-01 modules. Deep sleep requires a connection of GPIO16 to RST, that is hard to solder on the ESP-01.

1rabbit, 2018-01-09:
Yeah that would be awesome. I'll admit i was not thinking about that when i picked the esp-01, good point there.
I'm going to try to do it on one of theses and see how it goes, i see people have done it before, wish me luck with my 14$ iron

offtherails2010, 2018-01-10:
> Yeah that would be awesome. I'll admit i was not thinking about that when i picked the esp-01, good point there.
> I'm going to try to do it on one of theses and see how it goes, i see people have done it before, wish me luck with my 14$ iron

Yaa i have a cheapo soldering iron with standard tip sine brand new, its lasted me 9 years so far, and its my go-to iron for any and all soldering - Especially SMD, i hand solder upto 0402 sized SMD Components with magnifier, but can solder 0603 without magnifier :)

ESP-01 solder traces on the Actual IC are indeed difficult, fiddly and just downright annoying lol but it can be done without having to fork out for super expensive soldering tools, my soldering iron is a temperature controlled one, bought it for £20/$27 

This is why im going to be using the ESP07's for the indoor/outdoor wifi repeaters as they have deep sleep mode without messing about too much plus has option to add a proper high gain antenna instead of using the PCB Antenna, its a weird connector, called, U.FL to SMA Pigtail, then an SMA-Type High Gain Antenna to connect to the pigtail and pretty much got a higher reaching wifi repeater other than with just PCB Antenna (i will be testing this too with an ESP-07 without high gain antenna)

Just as an example to show its very possible to do so, i love SMD Soldering, if anyone needs a hand, lemmi know, be happy to help :)

martin-ger, 2018-01-10:
Have added the two parameters _am_scan_time_ and _am_sleep_time_ . After booting the esp_wifi_repeater scans for avilable uplink APs for _am_scan_time_ seconds. If none is found, it goes to deepsleep for _am_sleep_time_ seconds and tries again after reboot (default is 0 = disabled for both parameters).

1rabbit, 2018-01-10:
I ordered a couple esp-07 and some other kinds where gpio is easy to solder as well for further testings. In the meantime i managed to solder gpio16 to rst on the 4 esp-01 !

It took me a little while to get the technique right

[image]

[image]

[image]

[image]

martin-ger, 2018-01-10:
Fine work! Now, I hope you can use it...

1rabbit, 2018-01-10:
I'm trying it at the moment and it looks like its working fine with 1 repeater. Tomorrow im going to deploy them outdoors in a straight line and see how well it wakes up with 4 of them

offtherails2010, 2018-01-11:
@1rabbit 
well done with the soldering !!! very very nice job on them !! not so difficult eh :))

@martin-ger 
thanks to you once again for adding the sleep code in there as my primary intention was to use this wifi NAT Repeater as a standalone battery powered device, for the places that just need a little wifi boosting around the house - but then this will also help me in my other bigger project using sensor ESP's, Repeaters and a Raspi running MQTT Broker

**eeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeek !**

This is going to be my first time ever playing around with a Linux-Based main part of my build so im way out of my depth and comfort~zone lol 

hell of a~lot of reading needing & research.. wish i had gotten familiar with Linux 15 years ago, feel like old dog new tricks lol so a thanks in advance for any help i'll no doubt beg for along the way, thank you so very very much :)

1rabbit, 2018-01-11:
I'm back from my trials outdoors and there is a problem

In "lab" settings everything works fine :
Access point > The 4 repeaters approximately a meter away
When the access point is turned off, the 4 repeaters go into sleep mode at the same time, no issues.

In outdoors settings :
In a straight line separated by about 150 meters between each node:
Access Point > Repeater N°1 24:24:01:xx:xx:xx > Repeater N°2 24:24:02:xx:xx:xx > Repeater N°3 24:24:03:xx:xx:xx > Repeater N°4 24:24:04:xx:xx:xx

The repeating function works, but if i turn the access point off, the repeaters managed to keep each other awake without the Access point. They were flashing non stop and probably rebooting as well.

I had to get the 4 repeaters back in my pockets, turn on the Access point give them some time and turn it back off to get them back to sleep.

martin-ger, 2018-01-11:
Okay, think I know what is going on. Tried to disassociate STAs after AP loss now. Please check with the latest version.

1rabbit, 2018-01-13:
[image]

Great news, looks like the problem is solved from my testings.

I was a bit confused because when i flashed the new firmware the settings were staying the same on the esp-01 and not getting erased. Somehow it confused me and i thought flashing didn't worked so i wasted a bit of a time there. Then i figured its designed so you don't have to reconfigure at each update.

I ordered a 20$ food sealing machine to water proof the repeaters and some fishing line to get them in high spots. It should be there by monday so I can leave some of the repeaters outside and see how well they handle themselves over time.

offtherails2010, 2018-01-13:
Well Done 1rabbit - great job on that !

i was going to waterproof the little black project box for outdoor use but 2 things happened, 1 - i broke the lid's clamps to hold it in position lol and

2 - i forgot i had used up all my tiny batteries, i do have one from a tiny bluetoothh earpiece but seriously, it wouldnt last the hour lol

So... Solutions from here;; (btw, do you own a 3D Printer ?? with this kind of calibre stuff you do, id say its a no brainer that one should be furnished on your desk space lol)

1 - **[3D Printed Lipo 18650 Holder](https://www.thingiverse.com/thing:2019078)**, with space for adding/editing for space for the ESP-07, my ones will also be soldered to the wemos type boards so that i can simply hook up a USB Cable to iit to flash firmwares

2 - **[4 x 18650 Waterproof Enclosure](https://www.ebay.co.uk/itm/Waterproof-4x18650-Battery-Storage-Case-Box-Holder-For-Bike-LED-Light-Appli-D0G5-/282724535192?_trksid=p2349526.m4383.l4275.c10)**, not bad price actually - especially if you tear apart old dead laptop batteries - bound to be minimum 2-3 cells waiting to be rescued lol in this case, 4 x 18650's are all in PARALLEL - so all that juicy mAh being added across all batteries will make this particular enclosure last the longest outdoors

- also very nice idea for the fishing wire - im soooooooo stealing that one haha thanks man !!

martin-ger, 2018-01-13:
Great news! Thank you for your project and the tests.

I guess the easiest way to get some logging is to set up the central MQTT broker and use the build-in MQTT client. This can already report the Vdd and much other statistics. At least while the uplink is working...

1rabbit, 2018-01-13:
Yeah that's how i was thinking about implementing it originally. I even thought about triggering sleep from the MQTT broker before i suggested the default to sleep mode. It works a lot better with the internal sleep management now.

MQTT would be ideal to gather data about the voltage and store it to make some graphs etc.. In that case having a lot of data points matter. I will take some time to implement it and get everything automated and smooth. I just got an esp32 so maybe i could turn it into an AP + MQTT broker and use the extra power and space compared to the esp8266.

For total runtime and total sleeptime it would be handy to store the data locally, all we really need is the total so that's only two numbers to store. It would be reset manually for simplicity. We already have all the data locally (uptime + time to sleep)

I appreciate how you dedicated some of your time for this idea, this is promising.

martin-ger, 2018-01-13:
Unfortunatly, porting the stuff to the ESP32 is not absolutly trivial: Tasking under FreeRTOS is different, also the TCP/IP interface. Haven't even starting that so far.

For storing data locally during deep sleep there are only two options: either putting data into the RTC ram or flashing it. Too frequent flashing may damage the mem chips, RTC men has the disadvantage, that everything is lost after power down, but I will think about storing some statistics there...

1rabbit, 2018-01-13:
For the esp-32 i meant using it as an access point and MQTT broker with some code that is independent from the esp_wifi_repeater project. About the data storage, i guess you don't need to store the data while the esp is sleeping, you can do it while the esp8266 is awake and save it like the other settings right ?

CETCI, 2018-03-02:
This is great project and use for the ESP8266
I am at this time just connecting three ESP8266 - 12 first as Router soon to be Mesh
thank you

Instead of starting a new thread I thought this thread had the basics of what I was interested in doing and could be used by others using battery powered ESP8266.
What I would look to build was a line powered Route of a couple ESP8266 with at most three satellite Battery powered  ESP8266 using the deep sleep. The satellite's would wake every 30 sec. take a local read ( i.e Temperature ) if there was a delta change connect to Best AP then broadcast and go back to sleep if no delta change then sleep again. if no change after 5 Minutes broadcast a still alive . Now the question I would like to be able to not change the batteries for over a year - the problem is every time the STA wants to broadcast it has to establish a Wifi connection that takes time and power 
In Wifi there is a value of a Heartbeat of 100 ms to 1 sec is it possible for the AP to hold the connection open for 10 Minutes as well as have the STA store its AP information ( in RTC ) and wake up a send like there had never been a deep sleep
Understood that this probable breaks WiFi 802.11 but this is and end of line action and should have no effect on the rest of the LAN
In this project the system is a very static mesh so the AP/STA connection will rarely change
If one of the devices  ( AP ) is lost the STA should self heal as if the the AP had normally dropped 

If NO any thoughts on reducing connection times 
I will need the delta to be 30 sec and the still alive to be 5 minutes

martin-ger, 2018-03-02:
Happy to read that you like the project.

However: waking up every 30s is by far to much for the ESP to run for over a year: The ESP up and running needs about 70mA. Given that it runs only for 5 secs, which is not unrealistic for a wifi connect and send (also you need some time window for timing inaccuracy), you will get (even with a deep sleep current of 0A):
3600/30 = 120 wakeups per hour -> 120*5 = 600s runtime per hour -> 600/3600*70mA = 11mAh per hour. A good 18650 battery has about 3000mAh, meaning that it will run for about 3000/11 = 273 hours = 11.3 days...

CETCI, 2018-03-02:
Thank you for the response

sorry for misunderstanding 
I would not be having the ESP8266 doing the timing or readings. I will have a low power Micro doing the 30 sec readings and making the decisions when a broadcast is required.

The broadcast would on average only be every 5 Minute with only the middle broadcast if one of the 30 sec. wake ups determined the need

I need the 5 Minute for logging and proof of operation might be able to go 10 minutes if Delta readings can be guaranteed. 

The ESP8266 will be the Mesh device - this is why I am looking to see how much I can compress the communications time - as you indicate right now you are looking at 5 sec. 

The question is if the ESP8266's on both ends of the AP/STA connection could remember the connection for more than 5 Minutes - then the over head of setting up the connection could be bypassed and possibly reduce the connection time down to one second or less    
Continuous run current of 5 uA                                           3600x5uA          =  0.018 mAh
Wake every 30 sec current of 40 uA for 100 mS                  3600/30x40uA  = 0.0048 mAh
Transmit at 5 Minutes current of 70 mA for 1 second         3600/600x70mA   = 0.42 mAh
                                                                                               total                    0.4428 mAh
                              3000mAh/0.4428 = 6775 hr = 282 days
 
any reduction in transmit time is a big factor

offtherails2010, 2018-03-02:
A suggestion for outdoor battery powered devices like this - A [5v Solar Cheap chinese panel](https://www.aliexpress.com/item/Mini-0-15W-5V-Solar-Panel-Power-Panel-System-DIY-Battery-Cell-Charger-Module-Portable-Panneau/32813133390.html?ws_ab_test=searchweb0_0,searchweb201602_1_10152_10065_10151_10344_10068_10130_10324_10342_10547_10325_10343_10340_10548_10341_10192_10190_10084_10083_10618_10307_10301_10303_5711211_10313_10059_10184_10534_100031_10103_10627_10626_10624_10623_10622_5722411_10621_10620_5711311,searchweb201603_37,ppcSwitch_5&algo_expid=27958efa-e04b-4bf7-bc26-0440f2489df7-1&algo_pvid=27958efa-e04b-4bf7-bc26-0440f2489df7&transAbTest=ae803_5&priceBeautifyAB=0) with also a cheap [5v Lipo battery charger](https://www.aliexpress.com/item/TP4056-5v-1A-Lipo-Battery-Charging-Board-lithium-battery-Charger-Module-DIY-MICRO-USB-Port-30/32806599395.html?ws_ab_test=searchweb0_0,searchweb201602_1_10152_10065_10151_10344_10068_10130_10324_10342_10547_10325_10343_10340_10548_10341_10192_10190_10084_10083_10618_10307_10301_10303_5711211_10313_10059_10184_10534_100031_10103_10627_10626_10624_10623_10622_5722411_10621_10620_5711311,searchweb201603_37,ppcSwitch_5&algo_expid=8e803d72-6495-4453-8c9f-cb0c5b4403a1-0&algo_pvid=8e803d72-6495-4453-8c9f-cb0c5b4403a1&transAbTest=ae803_5&priceBeautifyAB=0), could charge up the lipo during the daylight hours and drain during non-daylight hours, making the module pretty-much self sufficient for not needing to be moved, charged or even touched,indefinitely, lol...... practically..

CETCI, 2018-03-02:
If only I had the sun on my back while doing this it would great :) - but alas I am inside a building - thus the need for Mesh - get in to all the little spots

CETCI, 2018-03-06:
Now have three ESP8266 in router mode - will setup in Mesh next and test dropping.
 
So the questions I have
Can the ESP8266 802.11 timing be changed - so If the 100ms Hartbeat is not answered for up to 5 Minutes the AP does not drop the connection.
When the ESP8266 goes into deep sleep can it retain the information required to re-establish the existing connection it had before the sleep i.e. started and send its information without having to go thru a reconnection.

martin-ger, 2018-03-06:
No, layer 2 connection handling is done in the closed driver code - no 
chance to change that standard behavior.

Also in deep sleep mode the RAM contents is lost. All state is lost. You 
can save several words in the RTC-ram, but certainly all WiFi and TCP 
connections are lost and have to bei newly established after restart (i.e. 
reset)