martin-ger/esp_wifi_repeater #126: Stream of garbage over serial after flashing ESP-12-F

jjg, 2018-01-12:
After following the Building and Flashing section of the README, the firmware flashed without complaint but when I reset the module (ESP-12-F) the blue light flashes rapidly.  If a serial terminal is attached an endless stream of garbage characters is displayed.

I tried a couple different baud rates to see if that was the problem but nothing seemed to work.  Is there a configuration or flashing parameter specific to this module that I may have overlooked?

marek26340, 2018-01-12:
You just missed one thing that might help you in README...

> Sometimes, especially on old ESP-01s, there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/espressif/ESP8266_NONOS_SDK/raw/master/bin/esp_init_data_default_v08.bin and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1).

jjg, 2018-01-12:
That did the trick, thank-you @marek26340 !

How did I miss that?

marek26340, 2018-01-12:
I don't know :D But happy to hear that! My ESP-12E is also doing this and always needs this .bin file that you have used now, then it works.

closed 2018-01-12 by jjg