martin-ger/esp_wifi_repeater #127: Use with vpn

sam-gurdus, 2018-01-12:
Is it possible to set this up so that all Internet traffic going through it is sent over a vpn?

sam-gurdus, 2018-01-12:
If not, does anyone know of an alternative? Thanks!

martin-ger, 2018-01-13:
Actually, it would be possible to tunnel the uplink traffic through e.g. a TCP-link. But I am not aware of any VPN-client for the ESP or the lwip. Probably PPTP would be a good candidate. While it is not the strongest encryption, it is probably of the size that would be realistic for the ESP.

I wont have the time to start a VPN client project, but I there is one, I will look into integration with my router.

sam-gurdus, 2018-01-14:
Yeah, for my use, I just want to use the vpn to bypass website blocks, so I suppose encryption isn't needed.

yunfan, 2018-10-09:
@martin-ger how about n2n , p2p vpn

@Glenta  any documents showing how to use this with proxy? i could have a socks5 proxy