martin-ger/esp_wifi_repeater #128: Very simple request

matthewkleinmann, 2018-01-17:
Howdy!  I love this project.  Thank you so much for taking the time to document it and make it public.  I have one minor request, and that is when you do any of the show commands, please list the parameters with the proper set values.  This makes is much easier if you want to change something and you don't have all the keywords memorized.  For example if you have a set command for auto_connect, make the show command preface the value with auto_connect and not autoconenct or auto connect.

martin-ger, 2018-01-17:
Did you try "help"? This is intended to give this overview.

matthewkleinmann, 2018-01-17:
Yes, help has all that info, but what would it hurt to use the same
keywords in the show commands as in the set commands?  I think that
consistency would be a good thing.