martin-ger/esp_wifi_repeater #132: Space character in SSID causes trouble

Muhamed-Reyad, 2018-01-21:
Hi im not sure why this is happening

Connecting to a network with a space in it's name yields a + character in the settings of the esp, and the esp then becomes unable to connect to the network.

When i created a similar network my self and replaced the space with + character the esp connected to it instead.

For reference The network ssid is "Eagle 0" , can anyone else verify if they too have ran into a similar problem ?

Muhamed-Reyad, 2018-01-21:
I have tried using hex encoding like %20 , but it also didn't work for some reason.

martin-ger, 2018-01-21:
On the Web Interface? Hex-encoding should work.
Try the CLI and Eagle\ 0

marek26340, 2018-01-22:
For a "+" you need to type in %2B, so if your network name is for ex. New+York, in the ESP, you need to type in "New%2BYork". If you have any other problems, feel free to ask here.

closed 2018-02-01 by martin-ger