martin-ger/esp_wifi_repeater #133: "MQTT cleint disconnected" message

impactful, 2018-01-25:
I am attempting to connect to a web MQTT broker, and I have set the parameters as follows at the "CMD>" prompt (below I Xed out user/pass info):

set mqtt_host m14.cloudmqtt.com
set mqtt_port 11278
set mqtt_user XXXXXXXX
set mqtt_password XXXXXXXXXX
set mqtt_id ESPRouter_2105b8
set mqtt_prefix none
set mqtt_command_topic test_topic
set mqtt_gpio_out_topic none
set mqtt_interval 15s
set mqtt_mask 0040

"show mqtt" command returns "MQTT client disconnected."  So it looks like MQTT is enabled, but the client is not trying (or not succeeding?) to connect to my broker.  I have tested the broker with another client publishing and subscribing, and it works OK.

I thought I might need to reboot ESP8266 to get the connection to occur, but reboot of ESP8266 just erases all the set MQTT values.

Any further suggested debugging steps?

Thanks!

martin-ger, 2018-01-25:
Yes, you need to reboot. Just enter all parameters, type "save" and reboot... Should connect then.

impactful, 2018-01-25:
I missed something simple:  I forgot to "save" config settings before "reset."  

After I save/reset, I get expected "MQTT client connected" response to "show mqtt."

closed 2018-01-25 by impactful