martin-ger/esp_wifi_repeater #136: ROUTE VS NAT

genoinosonda, 2018-02-01:
Hi Martin, how are you?
We are developing a project with IOT, and we would like to start making some contributions to the repository, in order to contribute to the community.
But before that we are discussing with our team the following purpose: How would a NAT mesh network with approximately 30 nodes? Using the ROUTE algorithm could not bring more efficiency in large networks?

martin-ger, 2018-02-01:
Interesting Question. What do you mean with "the ROUTE algorithm "? A special routing algorithm or IP routing in general?

The same kind of a tree that my "Automesh" creates would work with "normal" IP routing as well. But this would require a routing protocol on each node (e.g. RIP) and routing tables. The structure ot the network would be fairly the same (tree) as one ESP can always have just one uplink (STA). Both of these IP-based networks will only work well, if there is no (not much) mobility of the nodes.

The ESP drivers to not support IEEE 802.11s, so I think no chance to have a "real" WiFi mesh (if that makes sense with just one 2,4GHz radio anyway). To build a fully interconnected mesh you would need to use the (limited) raw send and receive capabilities of the ESP and would probably end up with something like the ESP-now from Espressif. This is not WiFi on layer 2, supports no TCP/IP and has significantly less bandwidth. But it can (probably) be more dynamic, less power consuming, allows for peer-ti-peer communication and has less setup latency. Maybe useful for special IoT applications with small sensor value as payload.

Did some work on Ad-hoc Mobile Networks: AODV is a recognized mesh routing protocol, but full implementation is really a challenge and performance in highly mobile environments still low.

I still think that my approach is well suited for a limited nuimber of mostly non-moving nodes (maybe 30 is already big) and an setup where all nodes in the end just want to be able to reach the internet. But I have to admit, I didn't do large scale test beyond 5 nodes so far... (Waiting for a number of 18650-battry powered ESPs)

ajanulis, 2018-05-17:
@martin-ger, did you get yours 18650 powered ESPs and tried large scale installation? I am interested in 10+ (up to 30)

martin-ger, 2018-05-17:
Yes, got them, but I didn't do tests with more than 10 nodes.

ajanulis, 2018-05-17:
@martin-ger, how much max did you you have in a single test? Somehow I do not feel very comfortable ordering 30 without certain level of confident

martin-ger, 2018-05-17:
I did a test with 8 nodes in a mesh - works.

The battery performance of the 18650 powered ESP boards is weak: their voltage converter burns a lot of power: with a cheap chinese 18650 with around 500-600 mAh a board like this: https://www.amazon.com/module-ESP8266-battery-Esp-wroom-02-motherboard/dp/B071H3JTXZ runs about 3h as repeater. A home-grown setup with 50ct charger circuit for over- and undercharging protection and a simple diode for voltage limitation makes it with the same battery for about 8h.

Also: on these boards, GPIO 16 is the blue LED.

ajanulis, 2018-05-17:
Ok, thank you very much @martin-ger . Have you ever considered to use LiFePo4 as it doesn't require any LDO and is "stress" tolerant for an abuse :) I am using LiFePo4 in 18650 form factor with 2 small (5V 100mA solar panels) and "ideal" diode to charge it - so far so good - almost a week (with about 3 hours direct sunlight) in 10 sec measuring 290sec deepsleep mode for ESP12 and constant power on BME280. GPIO16 for LED is not a good idea indeed :) Anyway, thank you very much for your amazing work and honestly prompt answers! Appreciate it