martin-ger/esp_wifi_repeater #137: Input voltage is not accurate in show stats

marek26340, 2018-02-02:
Hello again. This migh

martin-ger, 2018-02-02:
Actually, I am just using what the system_get_vdd33() function tells me...

There is some voodoo with the 107th byte of in esp_init_data_default.bin (named vdd33_const):
The SDK docs says:
Details of the [107] byte vdd33_const are as follows:
- If the [107] byte = 0XFF, VDD33 is measured internally, TOUT pin cannot work as ADC input.
- When the valid value range of the [107] byte vdd33_const is [18, 36], the unit is 0.1V, and the value should be set as the real power voltage of VDD33 to optimize the RF working condition. TOUT pin can work as ADC input.
- When the [107] byte vdd33_const is in the range of (0, 18) or (36, 255), the default 3.3V will be used as the power voltage to optimize the RF working condition. TOUT pin can work as ADC input.

In the default esp_init_data_default.bin that I have linked in the docs this value is set to 0x00. I tried to change it to 0xff. I don't really see a difference on my test Wemos D1...

The measurement does make sense anyway only if you are connected directly to the battery. Did that for an ESP01 board connected directly via a diode to a LiPo and here I can see the voltage going down over time.

If you have a converter in between (like on the Wemos or the NodeMCU), the system_get_vdd33() will always tell you the constant output of that converter. Then it would probably a good idea to leave [107] byte = 0X00, connect the ADC via a voltage divider to the battery and use system_adc_read() instead.

marek26340, 2018-02-03:
Thank you. 

Now I tried connecting a Li-Pol battery charged to 3.8V to the ESP8266 directly (no converter in between), did "show stats", showed me wrong voltage.. 
I tried to charge the battery fully to 4.2V, connected it to esp (directly!), did "show stats" and now the voltage is 2.4V.

What voltages are you seeing in "show stats"? Are they normal (4.2V~3.0V), or are they like mine (~2.7V)?

martin-ger, 2018-02-03:
On a plain ESP-01 I get 3,1 - 2,7 V slowly lowering when connected to a LiPo via a Si-diode: Somewhat realistic given that the diode drops around 0,7 V. On the Wemos D1 powered with the normal USB I get about 2,7 all the time.

Did you try the "107th Byte voodo"? Did you try measurements with other programs (e.g. Arduino code)?

marek26340, 2018-02-03:
I did not try that "107th byte voodoo". I don't even know how to edit that byte lol. But I could try to write that program which measures the input voltage.

marek26340, 2018-02-03:
And that 2.7V while powered via USB... I measured the output of my converter while powered via USB, and I measured perfect 3.3V.

martin-ger, 2018-02-03:
Added a version of esp_init_data with the FF set in the firmware directory.

marek26340, 2018-02-03:
I just made that file myself (0xFF on 107th byte) and you uploaded another one, that is completely the same. Got a hex editor, edited that 107th byte to 0xFF and uploaded it to the ESP. I'm sorry, but that didn't work. Not my .bin, not even your .bin.

martin-ger, 2018-02-03:
Any other result when you measure the voltage with another program?

marek26340, 2018-02-03:
Yeah, I made some code that reads the Vcc voltage and sends it to the serial monitor. The ESP was not transmitting any WiFi signals. I powered the ESP with USB, opened a serial monitor and it shown 2.990V. The real input voltage was once again 3.3V.