martin-ger/esp_wifi_repeater #138: Problems with automesh

philippd1, 2018-02-03:
Hi Martin,
I am trying to build up a mesh network out of a TP-Link Router with a hidden network and multiple D1 minis. Flashing the D1 minis works as expected and performance is perfect. But when I enable automesh (it does not matter if I do it via CLI or UI) the status LED just turns permanently on which according to the readme states that it is not connected to a AP...

Any help would be much appreciated.

philippd1, 2018-02-03:
I'm sorry, just messed up my config and didn't realize it^^

philippd1, 2018-02-04:
Sorry that I have to reopen this issue...

I configure my D1 minis like so (telnet):
> set ssid [SSID_router]
set password [PASSWORD_router]
set speed 160
set am_sleep_time 0
set am_scan_time 0
set ssid_hidden 1
save
quit
>
until this point, everything works fine but of course there is no mesh and the D1mini makes the wifi visible.

When I do this config via telnet, the D1mini flashes its light every second and after a reboot, it just just goes to a static light.
> set ssid [SSID_router]
set password [PASSWORD_router]
set speed 160
set am_sleep_time 0
set am_scan_time 0
set ssid_hidden 1
set automesh 1
save
quit
>

So...
Automesh is working when SSID is visible
Repeating a hidden SSID is working but makes it visible
**Creating a hidden mesh around a hidden SSID is not possible...**

Kind regards

martin-ger, 2018-02-04:
Okay, think, I can explain: Automesh does a manual scan to find the AP and the other nodes. In this scan it doesn't see the hidden SSID and thus it fails. Guess, the only workaround would be an intermediate ESP node as fixed level 1 router with a non-hidden SSID. After this, the other nodes will mesh again.

philippd1, 2018-02-04:
Okay, sounds logical. So this is the part "AUTOMESH_LEARNING" , is it?

Is there any chance that I could hardcode a hidden SSID with WPA2 password into the firmware?

martin-ger, 2018-02-04:
What exactly do you mean? The complete automesh hidden? That won't work. Need the scan for finding the possible/best route. The mechanism that allows for connecting to a hidden SSID is "Probe request/Probe response". These are management frames and the driver doesn't allow to use them in a user program.

You should be able to configure one (or more) ESP with the router firmware (without automesh enabled) to act as an intermediate between the hidded TP-Link AP and the automesh nodes. They may offer a different SSID than the TP-Link AP. So the TP-Link AP stays hidden and the automesh is visible.

philippd1, 2018-02-05:
Thanks for your response, yeah I wanted a complete automesh hidden. But your explanation makes perfect sense.
Maybe you could add this to the README in case others have the same thoughts as I did.

closed 2018-02-05 by philippd1