martin-ger/esp_wifi_repeater #139: feature request : Automesh - Startup as AP when the SSID is not found

LogoChris, 2018-02-05:
Good job on your software. I'm using the automesh mode to create a local network between multiple motocyles to communicate via a smartphone app.

Unfortunatly I have to create a hotspot with either another esp or the smartphone as root node of the mesh. 

Would it be possible to let the esp create an AP when the ssid is not around? So that there would be seperate networks if they loose each other but join into one mesh as soon as they are in range?

martin-ger, 2018-02-05:
Not so easy to add that in the generic software as each ESP that does not see a level 0 AP would begin to start its own mesh. So during boot-up many outer nodes might think, that there is no AP and will start.

Think, you would need one special alternative level 0 AP.

LogoChris, 2018-02-06:
I just noticed that in order to work, there is a difficult logic how and when to shutdown its own AP to join networks. 

Maybe a list of ssid:passwd with different priorities would do the job.
So every bike has another esp which acts as a level 0 AP but every bike with another ssid. 
And in the level 1-x esp every ssid is valid but there is a order of priority so if they are all in range they are in the same network.

Do you think something like this would work or is possible?