martin-ger/esp_wifi_repeater #141: uMQTT

chathurangawijetunge, 2018-02-11:
hi friend 
how to set the esp8266 as a mqtt broker...? where to set the broker port..?
can i use this as my worn stand alone MQTT broker..?

martin-ger, 2018-02-11:
You can: have a look at https://github.com/martin-ger/esp_mqtt

chathurangawijetunge, 2018-02-11:
there is no broker port set
only client port set is there

martin-ger, 2018-02-11:
Oh, you are right. You could change that in user_main, line 897 but I will add a CLI option "broker_port" later today.

chathurangawijetunge, 2018-02-11:
I'm having diffecaty in connecting to broker through my local rougher.
Can connect when I connect to my_ap

martin-ger, 2018-02-11:
No known issue here. Are you using the correct IP address? Can you "ping" it?

martin-ger, 2018-02-11:
Added "broker_port" now.

chathurangawijetunge, 2018-02-11:
hi friend

thanks for the update...

but still i can connect to the broker only when i connect to my_ap
i have got 192.168.1.4 ip from my rougher when i tried to connect the the
broker through my rough i cant

connected with SCW_*****, channel 11
dhcp client start...
connect to ssid SCW_ ***** ,channel 11
ip:192.168.1.4,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.4,mask:255.255.255.0,gw:192.168.1.1,dns:192.168.1.1
Got NTP server: 178.168.120.183
NTP synced
-------------------------------------------------------------------------------------
show
STA: SSID:SCW_ ***** PW: *****
AP:  SSID:MyAP PW:none [open] IP:192.168.4.1/24
Network console access on port 7777 (mode 3)
MQTT broker max. subscription: 30
MQTT broker max. retained messages: 30
MQTT broker username: admin
MQTT broker password: admin
MQTT client disabled
NTP server: 1.pool.ntp.org (interval: 300 s, tz: 0)
Clock speed: 80
Serial bitrate: 115200
CMD>

chathurangawijetunge, 2018-02-11:
and no i cant ping  ip:192.168.1.4

chathurangawijetunge, 2018-02-11:
its ok.... i think i have a ip conflict in my system i changed the ip to
192.168.1.45 now it works.

and if you can add DDNS update to like "www.dynu.com" so we can have our
won MQTT broker where we can access it from any ware in the world

closed 2018-02-18 by martin-ger