martin-ger/esp_wifi_repeater #142: Basic commands on programming

smyrnaff, 2018-02-11:
Hi everybody,
I am newbie on esp8266 and Nodemcu. For a long time, I am working on Arduino – Netduino but this is my first programming on Nodemcu. And now, I have a couple of questions about Nodemcu and Esp8266.

1. Can I use Esp8266 WiFi Module with “Esp WiFi Repeater” firmware (not Nodemcu SDK) on **Arduino**?
2. If it is possible, how can I use **basic commands** (like set, lock, reset etc.…) and **communicate** with each other?
3. When we use Nodemcu SDK, can we compile and run some code files (.**py** or .**lua**) on Nodemcu? Because I tried but I could not ...
4. As in the second question, how can I use basic commands during programming a .**py** or .**lua** file on Nodemcu SDK?

– Berk

martin-ger, 2018-02-11:
No, this is not a library that can be used in the Arduino or NodeMCU. It is a complete program that can be compiled with esp_open_sdk. If you want to extend the functionality, it is probably the easiest way to clone the repository and work with the code base. What exactly do you want to do?

It might be possible to assemble a setting that uses the patched lwip-lib https://github.com/martin-ger/esp-open-lwip in another build environment, but this is not a task for a newbie.

smyrnaff, 2018-02-11:
Actually, I tried to explain something like this;

While “esp_wifi_repeater” firmware is running on the Nodemcu, can I program and use GPIO pins of Nodemcu? As an example, I would like to display a data on a display led that I received with MQTT and this display have connected to Nodemcu. 

Or is it possible to communicate with Arduino via TX and RX after writing the “esp_wifi_repeater” firmware on the esp8266 (esp01) module? You said that it is not possible, but I mean that I want to use TX – RX of Arduino or Netduino (4 and 5 pins).

| GPIO View  | GPIO Connection With Adapter |
| ------------- | ------------- |
| [image]  | [image]  |

martin-ger, 2018-02-11:
You can alter the esp_wifi_repeater to do some aditional I/O, but this requires code changes, Currently it uses the TX/RX pins to do serial communiation with a connected terminal (the serial console).

smyrnaff, 2018-02-12:
Sorry for delayed time cause of timezone.

... And thank you for your information and response about all :) I started to working on it.

– Berk

closed 2018-04-01 by martin-ger