martin-ger/esp_wifi_repeater #143: Is there any way to expose internal network client directly outside?

WindyYam, 2018-02-12:
I'm connecting some devices using this repeater, that devices seems to sending udp packets to my phone's local address, not the nat's. I cant receive any of that packets from behind nat for that's an local ip. so are there any way to solve this?

martin-ger, 2018-02-12:
You have to use "portmap", this exposes a port from an internal device ar the repeater. You have to know the address of the intnernal device and the port it is listening to:

`portmap add [TCP|UDP] external_port internal_ip internal_port`
See also: https://github.com/martin-ger/esp_wifi_repeater#port-mapping

The easiest way is to map an internal port to the same external port. However, this can only be done once (for one device). If there are more than one device listening to the same port, you have to map them to different external ports and advise the sending APP to use them, e.g. with an URL like http:192.168.1.2:81/ if you are mapping an HTTP server.

WindyYam, 2018-02-12:
Thanks for reply but what i mean is that the device somehow sends packets to the internal ip.

e.g. The device opens an ap gw 192.168.10.1, the esp8266 opens 192.168.4.1,  i have my phone as 192.168.4.2, i have portmap assigned still no packet come and i can see from acl stats that many packets dropped(both ifaces are configured as allow any) so i infer the app send my local ip(192.168.4.2)to the device, the device send packets to this ip that canot reach

martin-ger, 2018-02-12:
Okay, the setup is:
APP <- 192.168.4.2 --- 192.168.4.1 -> ESP-AP | ESP-STA <- 192.192.10.x --- 192.168.10.1 -> quadcopter AP?

Think, I guess the porblem: The quadcopter receives the control packets from the APP (these open the NAP translation for this port TCP or UDP) and thinks it can send the video frames to the same address and some UDP port back. However the ESPs STA-side receives these packets and has no translation to the APP.
Guess, that a portmap on ESP with the UDP port used by the quadcopter for video to the 192.168.4.2 of the APP should work. Do you know this port no? Is it a fixed no?

WindyYam, 2018-02-12:
When I do a normal connection from my phone to my copter, I can see from my phone it listen to 2 UDP ports: 6525 and 6038. When doing with the repeater,  the 6525 port is ok since I can capture packets from my phone, but none of them are with port 6038 so i thinks this is the dedicated video port. In normal connection(without repeater), port 6038 only receives packets, port 6525 sends and receives. I have portmap these 2 port with "portmap add UDP 6xxx 192.168.4.2 6xxx". 

So my guess is the copter send video frame to phoneLocalAddr:6038, in my case it's 192.168.4.2 . If there's no repeater, that's ok but if behind the nat, it'll be dropped(correct me?)

anyway, this code is fantastic, I have test it on a DJI Mavic Pro and it works both control and video

martin-ger, 2018-02-13:
When looking into the code and thinking about it, I would expect that the packets are already discarded at the copter (it has probably no idea where to send a packet with a 192.168.4.0/24 address). If it reaches the repeater it should be forwarded corrctly. Any chance to verify that or perhaps to set a routing entry on the copter?

WindyYam, 2018-02-13:
yeah that's the main problem. if i have any idea how to print the packet information of the drone i maybe already figure out whats happening. i can see it says packet dropped after setting all acl interface to allow so it will.be good if i can print out those packet information

WindyYam, 2018-03-22:
I've done some research on those video packets.
I first try the normal setup Android -> drone and grab the transaction packets from adb interface. 
Then I tried the PC wlan -> drone setup, use some udp tool to send the correct transaction packets according to the grabbed file,  After the first transaction I get the correct local port 6038 video stream packets streaming to the PC, with most udp len = 1460. 
Then, I tried the PC->esp8266->drone setup, using the same method, portmapped, I get, in fact, a few port 6038 video stream packets from the uplink, but each of them udp len < 450. so I assume the packets size > 450 are abandoned.
Can you help me to confirm this? thanks

Err-Rahul, 2019-07-02:
Can you provide tutorial to implement NAT in esp8266. Thank you in advance!

martin-ger, 2019-07-02:
You need to replace the default lib lwip with this lib: https://github.com/martin-ger/esp-open-lwip#open-lwip-with-nat-slip-and-static-routing-for-the-esp8266
The Readme and the file https://github.com/martin-ger/esp-open-lwip/blob/enc_polling/include/lwip/lwip_napt.h explains the interfaces.

Err-Rahul, 2019-07-03:
I am unable to connect to 192.168.4.1
Getting error site can't be reached 192.168.4.1 refused to connect.
Why this error coming? How can I solve this?

martin-ger, 2019-07-03:
What does the serial console show?

UmutAlihan, 2023-04-24:
I think this issue is related with #430.