martin-ger/esp_wifi_repeater #145: http question

goosst, 2018-02-17:
Hello,

first of all, thank you for the hard work and making this program public.

I'm stumbling upon an issue: I have 1 Wemos D1 who works as a weather station and connects to my network to broadcast data through http (this works, I can check the data from my pc through http requests). I wanted to use a second Wemos D1 as a range extender of my wifi signal, this one runs your program.
So Wemos 2 runs your program and creates a subnet. Wemos 1 is connected to it and gets an ip address (10.24.1.3), I can successfully ping to Wemos 1 (through pc and through commands given to Wemos 2). However if I try to access by browser and make an http request I'm never able to connect to Wemos 1. I've played with a few settings of the configuration of esp_wifi_repeater (most likely not the correct ones) but wasn't able to make it work. Do you have any hints what could be blocking this?

thanks in advance!

martin-ger, 2018-02-17:
So the PC runs the browser and connects TO the Wemos (this is the server with port 80 open)?

If this is the case, you will need a port mapping:
portmap add TCP 80 192.168.4.3 80

This works, if your Wemos has the 192.168.4.3 in the repeater's network. To make sure, it receives always the same address, do a "save dhcp" once it has received it's address.

Now you connect your browser to the repeater's external address!

goosst, 2018-02-18:
thank you, works perfectly!

closed 2018-02-18 by martin-ger