martin-ger/esp_wifi_repeater #146: It is working but interfering with my TV signal

FBMinis, 2018-02-20:
I have successfully set up 3 wifi extenders using  Lolin NodeMCU V3 (ESP8266 12E) board, ESP-01 basic module and ESP-07S module. Thak you very much for sharing your work.
At home, we have a set-top box+phone+dsl system and, when the Repeater is ON, any channel I'm watching freezes every few seconds (I need to change channel for it to work again). What might be causing this sort of network-issue? It happens with the 3 Repeater versions I built.

Note: with the ESP-07S I had to download firmware_sdk_1.5.4 instead and use the Arduino Serial Monitor for setup (192.168.4.1 in the browser did not display the configuration window). I used the following circuit to flash but replaced the jumper with a momentary button (Click Download on ESP Flash Tool (Windows OS); Keep flash Button pressed; Press-release Reset button; Download begins, release flash button).
http://ediy.com.my/blog/item/133-upload-sketch-to-the-esp8266-esp-07-esp-12-using-arduino-ide

martin-ger, 2018-02-20:
So you are using IPTV? Your set-top box does not connect via the repeaters? Does it disturb your TV even when there is nothing connected to the ESP repeater? 

WIth this hint you will probably be able to run the latest firmware on the ESP-07: 

Sometimes there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/martin-ger/esp_wifi_repeater/raw/master/firmware/esp_init_data_default_v08_vdd33.bin and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1).

FBMinis, 2018-02-20:
My devices are connected as such:

https://conteudos.meo.pt/meo/UCMImages/CustomerCare/2013/ESQUEMA-Ligacoes-LigacoesMeoADSL.jpg

My laptop and smartphone are usually connected to the router by Wifi. The device named "Meobox" in the picture enables the TV to explore, display and record the subscribed channels. I only use one Repeater. It is placed in my room, which is away from the router and I only connect my tablet to it.

The channels displayed in the TV freeze even when there are no devices connected to the Repeater (just turned ON). The intervals at which the channels freeze dont seem to present a time pattern, neither does how long the freeze lasts.

martin-ger, 2018-02-20:
So the Meobox is NOT connected via WiFi - this means, RF-interference is probably not be the issue. 

Strange, if you just start an ESP with another program that:
a) connects as client to the router or
b) opens an AP on the same channel as the router
Any issues here?

martin-ger, 2018-02-20:
To check whether there is any logical interference let's try the following:

Install the latest firmware and enter at the CLI:

set acl_debug 1
acl from_ap IP any any deny

This should deny any interaction but prints whatever the repeater "sees"...

FBMinis, 2018-02-20:
I installed the latest firmware, opened the Serial Monitor of the Arduino IDE (only one Im familiar with), then typed "set acl_debug 1" and clicked Send, also for "acl from_ap IP any any deny". 

Then I went to the config window and set my STA settings. The channel started freezing.

martin-ger, 2018-02-20:
Do you see any Log output on the Serial Monitor?

FBMinis, 2018-02-20:
I am not sure if I'm sharing sensitive information. 

`WiFi Repeater V1.6.4 starting

Config found and loaded
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(68:c6:3a:85:81:54) + softAP(6a:c6:3a:85:81:54)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with Reis, channel 1
dhcp client start...
connect to ssid Reis, channel 1
ip:192.168.1.66,mask:255.255.255.0,gw:192.168.1.254
ip:192.168.1.66,mask:255.255.255.0,gw:192.168.1.254,dns:192.168.1.254
set acl_debug 1
ACL debug set
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>acl from_ap IP any any deny
ACL added
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.254:25621 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:25621 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:25621 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:25621 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:25621 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:25621 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:25621 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:25621 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.254:1055 Dst: 239.255.255.250:1900
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>
deny: UDP Src: 192.168.1.64:1044 Dst: 239.255.255.250:8082
CMD>`

martin-ger, 2018-02-20:
Nothing special here: some UPnP stuff, guess 192.168.1.64 is your Meobox. Don't think there is any network interaction that explains the problems...

If you start it and then do:
set ap_on 0
still the same? This just disables the AP...

FBMinis, 2018-02-20:
> set ap_on 0
bcn 0
del if1
pm open,type:2 0
mode : sta(68:c6:3a:85:81:54)
AP off
CMD>

The channel freezes! : )

martin-ger, 2018-02-20:
set ap_on 1
disconnect
This cuts the STA connection... ?

FBMinis, 2018-02-21:
Sorry for the delay, regular life.

There are freezes even after "disconnect" through the terminal. 

Installing the WifiClient sketch from the Arduino IDE does not generate freezes (if I connect to the router or not).

Installing your latest firmware and pressing reset, MyAP shows up. If I dont input the settings and connect to the STA, no freezes. If I input the setting and connect, the freezes begin.

marek26340, 2018-02-22:
Are you setting the higher CPU speed? ("set speed 160")

FBMinis, 2018-02-22:
set speed [80|160]

closed 2018-03-05 by FBMinis