martin-ger/esp_wifi_repeater #147: Sparkfun - Thing

PaBz0r, 2018-02-25:
Hi, I am struggling to get it working. I am using the ESP Tool with the following settings:

[image]

On the serial monitor I can see the following (BAUD 74880):
[image]

Nothing appears on wifi searches and I can't execute any commands. 

Pretty new to this, excuse me if I am missing something obvious :)

martin-ger, 2018-02-25:
Looks not so bad. Baudrate should be 115200...
Is is really just only 512 kB of Flash?
Try SPI speed 40MHz

PaBz0r, 2018-02-26:
I just see jibberish on 115200. Will try 40MHz, thanks. Yes by the looks of it only 512kb. I found an older version 1.5.4 which worked first time (instructables article). Only difference seems to be flashing to 0x40000, not sure what that has to do with it.

martin-ger, 2018-02-26:
Then it is most probably a problem with esp_init_data. The ESP tries to find them 4 sectors below end of flash. When the old binary (with the old esp_init_data and address 0x40000) works and the new doesn't even when you tried to flash it with the new esp_init_data, then most probably you flashed it to the wrong sector. Please double check the size of your flash or - just in case - try to flash it to 0xfc000 (1MB) and 0x3fc000 (4MB) as well...

marek26340, 2018-02-28:
And also why is he flashing that vdd33 esp_init_data that you made for me?

martin-ger, 2018-02-28:
Why flashing the esp_init_data_default_v08_vdd33.bin from https://github.com/martin-ger/esp_wifi_repeater/blob/master/firmware/esp_init_data_default_v08_vdd33.bin?

The SDK need certain config vars in that place. This data structure has changed since the SDK 2.0. If there was an older firmware on the ESP before, starting a program compiled with the new SDK fails during init, when the old structure is in place.

The "vdd" verson is just the one that uses the ADC for measurement of the chips power supply, not an external source. (107th byte in esp_init_data_default.bin, named as vdd33_const, determines that).

marek26340, 2018-02-28:
ok then, it doesn't matter what will be flashed

martin-ger, 2018-02-28:
As long as it is from the latest SDK - no. The one from here is find as well: https://github.com/espressif/ESP8266_NONOS_SDK (but Vdd measurement will not work).

PaBz0r, 2018-02-28:
I have tried it a couple of times with the same result.  I can also confirm that my flash size is only 512kb.  Is there anything else I need to flash for SDK 2.0, as per the NONOS_SDK link above? 

Thanks for the help!

closed 2020-08-04 by PaBz0r