martin-ger/esp_wifi_repeater #148: Very Unstable AP (fluctuates)

ThomasQvist, 2018-02-28:
Hi Martin 
Thank you for a fantastic project!.
I am having a problem a Wemos D1 mini pro board (https://wiki.wemos.cc/products:d1:d1_mini_pro)
I have flashed your firmware on the board (version 1.6.4), and powers the board via the USB. (connected to a PC with Arduino for console).

When observing the wifi-signal using a tool like Vistumbler - I can see the the signal of the access point fluctuates a lot (or you could say - it cuts of every 5 seconds or so)
[image]

I do not get any error messages (or any other diagnostics) via the console (serial monitor connected via the USB) when the ap is cut off.
I have also tried to power the board via an usb-power supply with the same pattern (so I think that my PC delivers enough power).

I have tried to connect the board to the internet - and I can (without missing a single package) send ping 8.8.8.8 commands, while I observe (via Vistumbler) that the AP cuts off - So I don't think that the board goes into deep sleep.

I really hope you can help me, since this project is just what I have been looking for.

Best regards
Thomas Qvist

martin-ger, 2018-02-28:
Don't know about a problem like this. Is it really a functional problem or just an artefact of the measurement - meaning: does a client really experience any problems when connected via the AP?

ThomasQvist, 2018-02-28:
The problem is real - it is practically impossible for me to connect to the AP (having tried many different devices) - I also see the exact same pattern on my Android Phone (using Wifi Analyzer).
If I am lucky I get connected, and is then thrown off seconds later.
I have just tested it with the new 1.6.5 firmware (also making a 'reset factory') - and the problem is the same.

I have also tried the firmware on a standard Wemos D1 mini (https://www.aliexpress.com/item/D1-mini-Mini-NodeMcu-4M-bytes-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266-by/32644199530.html) and here I see the same pattern, but the disconnects come a little less often (approximately every 10-15 seconds).
[image]
This makes it easier to connect, but the connection is still quite unstable.

I bought the D1 mini pro especially for your firmware (since it has an external antenna).

(not related to this issue, but the voltage in 'show stats' for version 1.6.5 now shows 'Power supply: 63.996 V' - not a problem - know that the voltage should be used for battery use - but a little confusing ;-)).

martin-ger, 2018-02-28:
Please try this procedure:

Sometimes there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/martin-ger/esp_wifi_repeater/raw/master/firmware/esp_init_data_default_v08_vdd33.bin and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1).

ThomasQvist, 2018-02-28:
Sorry - I have tried that procedure (in advance - followed some of your suggesting on the other support-threads).
I have, however, just tried it again with no luck - just in case ;-)

Please note that the software is running on the board - I can connect via the serial monitor on the Arduino IDE, and I can execute commands via the command line interface (such as ping and 'show statsø or connect to my wifi via 'set ssid' and 'set password').

PS. Wow - you are so fast answering these posts - thank you for your dedication 🥇

martin-ger, 2018-02-28:
And the Power supply issue remains? That's wired...

Perhaps the problem arises as you Wemos has 16MB flash? It always look for the sector in the last 4 sectors of flash.

Try to flash the init-sector to 0xffc000.

ThomasQvist, 2018-02-28:
Flashed init-sector to 0xffc000 - same pattern :-(

martin-ger, 2018-02-28:
At least the power supply issue solved (indication that the init_data is in the correct place)?

Other ESP programms with AP (same SSID) don't have these problems?

Is there anything special in your WiFi environment? Wireless IDS or something like that? another ap_ssid doesn't help anything?

ThomasQvist, 2018-02-28:
I will try it in another environment - I have a lot of wifi signal here - both from my Velop mesh routers and several neighbors - I do however see the same pattern whether or not I am connected to an ap from the esp8266.

But you are right about the power supply now shows the right voltage - so one problem solved - just one to go.
If you (or other) experience the same problem, or have any suggestions to a fix - please write here.

I will receive a new batch of Wemos D1 mini pros in a couple of weeks - then I can verify whether it is a hardware issue.
Thanks for your help

ThomasQvist, 2018-02-28:
changed the title to reflect that it is only (as far as i know) an issue for a Wemos D1 mini _**pro**_

gr01d, 2018-07-22:
Confirming that ESP WiFi Repeater works on **Wemos D1 mini**

mjf55, 2019-06-02:
Any update on this issue?  Is the D1 Mini Pro still an issue with this software.  

ThomasQvist; did you try with the new hardware and does this still exit?

Thanks