martin-ger/esp_wifi_repeater #149: Feature request: Mesh map

HLNZ, 2018-03-02:
Hi Martin,
Would it be possible to have a command (meshmap?) that returns:
-ip adress of unit
-what station the unit is connected to & signal strength (ie upstream connection-like the scan command)
-connected stations & signal strengths (ie downstream connections)
And then issues the same command to all the units connected downstream, and gets the results passed back to all the units connected upstream.

This is for working out and sorting signal issues onsite, without having to go to each unit in turn.
It should hopefully allow a heatmap (of signal strength) of the various stations and the wifi connections.
Also hopefully, this would also show where further repeater nodes should be placed.

Regards,
Robert

martin-ger, 2018-03-02:
Distributing some infos on the structure of a mesh is really a good idea.

Getting the RSSI info of downlink STAs isn't easy (possible), I don't know an API for that. :-(

My suggestion would be to send a JSON struct with the following infos to some central MQTT broker:
NodeInfo: user defined name, AP MAC, STA MAC, Uplink RSSI, no of downlink STAs
For each downlink STA: STA MAC,IP

This would be enough info to construct a fully attributed graph of the mesh and it could be implemented quite easily with the already existing mechanisms.

martin-ger, 2018-03-03:
Just added a new MQTT Topic that is published regulary:

The "TopologyInfo" topic contains the following JSON structure, that can be used to reconstruct a complete graph of an automesh network:

```
{
"nodeinfo" {
	"id":"ESP_07e37e",
	"ap_mac":"ee:7c:87:07:e3:7e",
	"sta_mac":"ec:fa:bc:07:e3:7e",
	"uplink_bssid":"00:1a:54:93:23:0a",
	"ap_ip":"192.168.4.1",
	"sta_ip":"192.168.178.33",
	"rssi":"-66","no_stas":"2"
},
"stas":[
	{"mac:":"5c:cf:45:11:7f:13","ip":"192.168.4.2"},
	{"mac:":"00:14:22:76:99:c5","ip":"192.168.4.3"}
]
}

closed 2018-03-09 by martin-ger