martin-ger/esp_wifi_repeater #150: Create Static Routes

cnagano, 2018-03-02:
Hi Martin!

I was wondering if it is possible to add static routes to ESP. The thing is, I've managed to use it as a wifi repeater, but only the first ESP (and his nodes) can be seen on the network.

laptop1 <--> router() <--> (sta)ESP(ap) <--> laptop2   THIS WORKS(adding a static route on the router)

laptop1 <--> router() <--> (sta)ESP(ap) <--> (sta)ESP(ap) <--> laptop2 
On this last scenario laptop1 cannot see laptop2 and I think a static route could solve it. I saw that your FW allows to portforward, so I'm wondering if it could add a static route as well...

Thanks in advance!
Regards!

martin-ger, 2018-03-02:
Currently, all AP interfaces are NATed. If you want to forward single port,you can do this by configuring two portmaps, one on each ESP, e.g. on the first: port 80 to port 80 of the second ESP, on the second: port 80 to port 80 of laptop2. Not quit convenient, but it works, as long as you have a small number of known ports you know the IPs.

I am just working on static routing for lwip (see https://github.com/martin-ger/esp-open-lwip), but I do not fully understand how this interferes with NAT (probably you would have to disable it).

Will release a version with an "add_route" command soon...

cnagano, 2018-03-02:
Nice!! Im trying to merge all subnets of all ESPs and without the static route everything only goes from children to parent... 
Looking forward to it
Im gonna check this project of open lwip also!
Thanks again!!

algood, 2018-03-30:
**Route** is performed through (at least) ip.c and etharp.c (from LwIP package).
Testing icmp (ping) shows that there is a limitation in the following scenario:

PC03 - AP[esp03]STA - AP[esp02]STA - AP[esp01]STA - router - internet
-------------------- PC02 ---------- PC01 -----------------------

Routes are set on esp to define networks seen on their left hand only (AP side and through other esp).

**Ping is ok for:**

- PC03 to any (esp, PC, router)
- PC02 to any except destinations on the left of esp03 (including AP03)
- PC01 to any except on the left of esp02
- any esp to any

The PC are set by DHCP with GW=APesp. Eg: PC02 has GW=APesp02.

From this PC02, we can reach APesp03 or PC03 only by adding a route on PC02 with GW=STAesp03.
This is not generic since only routers shall have knowledge of routes.
Something is maybe missing in etharp.c.
**_Is anyone meeting this limitation ?_**

algood, 2018-05-15:
**Investigation statement of ping/route issue**
Modifications in **ip.c** is a solution: 
Taking care of source and destination address in ip_header when choosing the next hop address is one part of the solution to recover full capability of ping.

Use of NAPT in case of ICMP payload has to be analyzed further.

Maybe too many call to ip_route function in a complete packet processing per ESP ? Eg: in raw_sendto, in ip_forward, in output_if => to be checked wether required.