martin-ger/esp_wifi_repeater #152: How you can set max 8 station for one AP esp8266?

tinamore, 2018-03-03:
Hi,
You have written in readme:
```
Due to the limitations of the ESP's SoftAP implementation, there is a maximum of 8 simultaniously connected stations.
```
But i see in document esp8266 arduino:
```
maximum number of stations that may be connected to ESP8266 soft-AP is 5.
```
and I was known some info from google tell maximum station for AP as 4

How you can set max 8 station for one AP esp8266?

martin-ger, 2018-03-03:
With "wifi_softap_set_config()" you can set the confg, "struct softap_config" has a component "max_connection". 

Docu says 4, but it does accept values up to 8, when setting it to a larger val, setting doesn't work. I think mem is the limiting factor...

martin-ger, 2018-03-03:
Please try to print out the result of system_get_free_heap_size() - is it the mem?

Does this happen with WPA2 encryption or with an open network?

tinamore, 2018-03-03:
Hi,
I have use project https://gitlab.com/BlackEdder/painlessMesh for test.
I see this painlessMesh have same your config max_connection in src/painlessMeshAP.cpp
```
 apConfig.ap.channel = _meshChannel;
    apConfig.ap.beacon_interval = 100;
    apConfig.ap.max_connection = _meshMaxConn; // how many stations can connect to ESP8266 softAP at most, max is 4
 esp_wifi_set_config(ESP_IF_WIFI_AP, &apConfig);// Set ESP8266 softap config .
```
I try set apConfig.ap.max_connection = 8.

FIRST: i created one AP_ONLY with max_connection = 8
AFTER: I create 8 node with mode STA_ONLY
THEN: I print list node connected to AP  node. I see node list always = 5.

Mesh use WPA2 encryption.

martin-ger, 2018-03-03:
I am currently connected to my repeater with 6 STAs (4 ESPs, a smartphone, and a kaptop) with WPA2 - works and still 12KB left...
```
CMD>show stats
System uptime: 0:35:58
Power supply: 2.922 V
716 KiB in (6225 packets)
3410 KiB out (6393 packets)
Phy mode: n
Free mem: 12560
External IP-address: 192.168.178.33
6 Stations connected to SoftAP
Station: xx:xx:94:75:54:0c - 192.168.4.5
Station: xx:xx:e8:4e:42:c5 - 192.168.4.7
Station: xx:xx:7f:11:7f:13 - 192.168.4.2
Station: xx:xx:22:76:99:c5 - 192.168.4.4
Station: xx:xx:d6:77:45:22 - 192.168.4.3
Station: xx:xx:7f:d8:bc:b5 - 192.168.4.8
```

tinamore, 2018-03-04:
Ok, thanks.
May be your esp_wifi_repeater use native code, high performance.

closed 2018-03-04 by tinamore