martin-ger/esp_wifi_repeater #154: Spaces don't work in WIFI station names

jcea, 2018-03-05:
I just tried to "join" a network with a space in the WIFI network name. That space, using the web configuration interface, is replaced with a "+", following HTTP specs. This project should replace "+" as "space" internally.

Currently, I can NOT join this network, with spaces in the name.

Thanks!.

martin-ger, 2018-03-06:
On the CLI use '\ " instead.

marek26340, 2018-03-07:
I remember that only "\ " and "%20" make a space...

martin-ger, 2018-03-09:
If you want to enter non-ASCII or special characters in the web interface you have to use HTTP-style hex encoding like "My%20AccessPoint". This will result in a string "My AccessPoint". With this hex encoding you can enter any byte value you like, except for 0 (for C-internal reasons).

closed 2018-03-09 by martin-ger