martin-ger/esp_wifi_repeater #159: NodeMCU not waking up from deep sleep normally

marek26340, 2018-03-11:
Hello

I've got a NodeMCU v2 doit.am with ESP12E onboard.
When I put my NodeMCU to deep sleep with command "sleep 10" (in telnet), well, it just goes to deep sleep. That's normal. But when it's time to wake up, it just sets the GPIO16 (low/high idk) which doesn't wake my ESP up, but only turns on the second onboard LED. I even tried to set the LED GPIOs manually. Set to 2, the blue LED on the ESP8266 lights up. I then set it to pin 16 and the blue LED on the NodeMCU board lighted up. I previously had a NodeMCU v3 board that did not had this issue.
What should I do?

Here's the serial output at baudrate 115200:
`Config found and loaded
Starting Console TCP Server on port 23
mode : sta(60:**:**:**:**:**) + softAP(62:**:**:**:**:**)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with e***********a, channel 11
dhcp client start...
connect to ssid e***********a, bssid 50:**:**:**:**:**, channel 11
ip:192.168.*.**,mask:255.255.255.0,gw:192.168.*.*
ip:192.168.*.**,mask:255.255.255.0,gw:192.168.*.*,dns:8.8.8.8
add 1
aid 1
station: a4:**:**:**:**:** join, AID = 1
station: a4:**:**:**:**:** join, AID = 1
tcp_client_connected_cb(): Client connected
state: 5 -> 0 (0)
rm 0
del if0
station: a4:**:**:**:**:** leave, AID = 1
rm 1
bcn 0
del if1
usl
disconnect from ssid e***********a, reason 8
station: a4:71:74:3d:75:62 leave, AID = 1
enter deep sleep`

And here is serial output at 74880 baud when the ESP wakes up:
`ets Jan  8 2013,rst cause:5, boot mode:(3,7)

ets_main.c`

What? Now we see!! The ESP wakes up but doesn't execute the code!

martin-ger, 2018-03-11:
Did you connect GPIO16 and reset? This is mandatory, If you want to use 
deep sleep.

marek26340, 2018-03-12:
Looks like on my old NodeMCU the GPIO16 was connected to RST. My new one doesn't have that connection. Okay then, closing issue now.

closed 2018-03-12 by marek26340