martin-ger/esp_wifi_repeater #160: Feature request: Option for random MAC address change for STA on every re-connect

rokrodic, 2018-03-11:
Hi. Simple request. Can you add an option to setup page (HTTP) if one wants the ESP's MAC address to change on every renowal of connection. Some free-wifi-sharers block reconnection of the same device (MAC blocker). I.E. if ESP cannot reconnect in let's say 5 attempts, it would change its MAC address and retry...
Sounds a good idea?

marek26340, 2018-03-12:
Stealing a internet connection might not be illegal, but it just isn't good. Just google that if stealing WiFi is illegal in your country. Also if it is a free open WiFi with no security, you are in a risk of loosing data etc............................

martin-ger, 2018-03-12:
Using a random MAC is a legitime privacy issue (and probably technically no big deal). Will think about it for the next release.

Whether or not sharing an authenticated access is allowed is probably beyond this discussion. Actually any NAT router allows for that and even any Linux box can be configured to do so.

rokrodic, 2018-03-12:
I opted for feature upgrade. To broaden the spectrum of features of this nice piece of SW Martin offers...
But to clarify: It is illegal if used in bad way. In my case I get internet from my neighbour who allows my usage, but for the router he uses he does not know the pass neither he allows me to factory reset it. For now I use my laptop - Windows 10 allows random MAC!!! 
His router is set in an odd way. It allows 1GB transfer then the MAC is blocked. I do not advise illegal usage. Who uses, takes risk. And it is beyond this discussion as Martin said. Sorry for my clarification.

marek26340, 2018-03-12:
Does this happen to your neighbor too? (transfer 1GB, then blocked)

marek26340, 2018-03-12:
Does this 1GB limit apply to your neighbor too?

And also, sorry then :D no stealing of wifi

rokrodic, 2018-03-12:
Nope. Unlimited GBs. Just speed is limited to 20Mbps...

marek26340, 2018-03-12:
So your neighbor has no limits but the router will disconnect you if you surf over 1gb. isn't your neighbor hiding something from you? :o

rokrodic, 2018-03-12:
Maybe. As I pay half of the internet bill, it would be nice for both of us to have unlimited internet. But my neighbour is 80 years old, half deaf and barely walks. He resists any changes. Respect! I look at it: he is a nice person and we are good neighbours and we share some of the costs.

closed 2018-03-20 by rokrodic

martin-ger, 2018-04-01:
Try "set sta_mac random"

This will set a new random MAC on each reboot.